gereic/GXAirCom #135: "Broadcast weather-underground data" needs https

bzed, 2023-07-03:
With 49ca39daa3498ef97f95caa003fa5f49d05b02a5 https was disabled for weather underground, which works fine for uploading data - but the broadcast WU data feature needs https as WU sends a 301 redirect to https on the API url.

gereic, 2023-07-03:
problem is, that https is only working with wifi.
On Sim7000 it is not working, because only 2 connections are allowed.
I'm currently working on another solution, that the Ground-station is communication only with mqtt to a server and the server is doing all connections to WU, WI, ...

bzed, 2023-07-03:
mqtt sounds like a good option, I thought about doing something like that as well, but didn't find the time yet. Also I was looking into feeding the mqtt data into telegraf, but the format is not what telegraf is expecting, it might make sense to think about the json format in the mqtt message to make it easier to use telgraf to process them. Especially as this would provide a very easy way to write the data into basically all other kinds of databases.

gereic, 2023-07-03:
at the moment, I write the weather-data to the topic GXAircom/_DEVID_/WD
for example:
{
  "DT": "2023-07-03T12:45:53+00:00",
  "wDir": "227.01",
  "vaneValue": "225",
  "error": "0",
  "wSpeed": "4.27",
  "wGust": "9.36",
  "temp": "29.87",
  "hum": "51.89",
  "press": "1015.56"
}

My Server gets the data with a simple python-script an writes it into a mysql-database.
Also, it sends the data directly to windy/WU, whatever.

With a simple php-script, I'm getting out all data for example wind.mobi
[http://gxaircom.net/gxaircom/stations.php](http://gxaircom.net/gxaircom/stations.php)
[http://gxaircom.net/gxaircom/stationstable.php](http://gxaircom.net/gxaircom/stationstable.php)

bzed, 2023-07-03:
Yes, the python script version is easy for us, although I think for the common user it might be easier to support telegraf out of the box as it would allow to collect the data and write it to influxdb or any other DB without having to code anything. (The current json format should work with telegraf, but it will need a it of config to parse it properly).

I can look into it if its okay for you, or share a telegraf config for the old version...

gereic, 2023-07-03:
The idea behind is, to setup a webpage, where you can configure all things for the ground-station and you have nothing to adjust on the ground-station itself.
Also the code for the ground-station will be much easier.
Do you know, if telegraf can also send the weather-data to WU/WI ?

bzed, 2023-07-30:
I think uploading to WU could be possible with the http output plugin of telegraf and the json output format. Not tested at all though.

Regarding boardcasting WU data: I think one good option would be to use mqtt. Subscribing to some weather station data and boardcasting it would avoid an extra connection, the mqtt connection is open already. Needs some server to feed the data, but as long as its documented well it should not be hard to provide.

For those who are not able to run their own mqtt server, I wouldn't mind to run a public service to feed weather data from WU and maybe holfuy and other sources. Just no idea when I find the time to code it...