gereic/GXAirCom #136: getFloat vs putUChar in fileOps.cpp

jccricket, 2023-07-04:
Hello,

I have a question about fileOps.cpp
Why is:
` preferences.putUChar("WDANEOADSVDIVR1",pSetting->wd.anemometer.AnemometerAdsVDivR1);`
 but
` pSetting->wd.anemometer.AnemometerAdsVDivR1 = preferences.getFloat("WDANEOADSVDIVR1",8060000.0);`

(same for R2)

Harmonising it removes an error at runtime.
There is still the BATOFFS that is not "written" in the preferences.
Indeed, the logic is not the same as for other parameters and it uses a function, which is never called except if you go to the webinterface.
'void write_battOffset(void)'

I would be happy to see a BATOFFS written to 0 (or any other value that make sense) by default.

And maybe a general comment.
If ESP32 flash memory is erased, and GXAirCom run first time, all those parameters trigger errors.
Is there an option to have a "default" setting file written at first launch of the app ? (or even before the first launch, at flash time)

Please consider I am a novice in the area. And maybe my focus is more on making the code more "readable" for someone like me who is joining the project now and have some difficulties to understand it all.

if that make sense, I could share later similar findings. - I would like to focus on board detection, but I don't have different board at home to make tests, so maybe not the best. I can still continue my investigation in the AXP202_X area, since batterie is not working with my board version ESP32-D0WDQ6-V3 (revision v3.1)

bzed, 2023-07-07:
Hi,

> I have a question about fileOps.cpp Why is: ` preferences.putUChar("WDANEOADSVDIVR1",pSetting->wd.anemometer.AnemometerAdsVDivR1);` but ` pSetting->wd.anemometer.AnemometerAdsVDivR1 = preferences.getFloat("WDANEOADSVDIVR1",8060000.0);`
> 
> (same for R2)

because I failed in copy&paste :)
I'll send a PR to fix this one.

Thanks for catching it,
Bernd

gereic, 2023-07-07:
Thanks for the hint.
It will be fixed in next version.

jccricket, 2023-07-08:
I will remove from the PR I did for AXP2120.
Indeed, I'm learning and I think that PR that I did can be improved a lot. And probably a better idea to not mix everything. 

JC

gereic, 2023-07-14:
fixed in version v5.3.5

closed 2023-07-14 by gereic