gereic/GXAirCom #137: TBeam-AXP2101-V1.2 doesn't work with current AXP implementation

jccricket, 2023-07-05:
Hello,

I have a TBeam-AXP2101-V1.2 and unfortunately, the current AXP implementation doesn't work. (Indeed, AXP2101 is NOT AXP192).
I have found library XPowersLib (https://github.com/lewisxhe/XPowersLib) which is developped by same person who did AXP202 library.

I tried to implement the XPowerLibInterface, but this was not following GXAirCom current structure, so, I decided to add everything in "main.cpp" despite this is not my preferred best practice.

I did implement for myself, but think this is worth sharing as this new XPowersLib sounds quite good in how to manage battery.
Something like this sounds more readable to me in case the board has a PMU:
`bool printBattVoltage(uint32_t tAct){
  #ifdef BOARD_HAS_PMU
  if (!PMU){
    log_e("PMU not initialized");
  }else{
      status.vBatt = PMU->getBattVoltage();
      status.BattPerc = PMU->getBatteryPercent();
      return true;
  }
  #endif`
  
  
This is my first contribution, and I'm pretty sure this is not a good idea to merge. Anyone who would like to guide me and make sure what I'm introducing is not going to brake anything else, please feel free to reach out. I'm more than happy to invest time in this project.

Now, I need to learn how to make a pull request with GIT...

gereic, 2023-07-16:
updated GXAircom to XPowerLib.
Works on T-Beam (axp192) and S3Core (axp2101).
Can you test it ?

jccricket, 2023-07-17:
Hello,

I had a bit of time before work this morning and I did a quick build, upload and monitor.

[I][main.cpp:1126] setupPMU(): AXP2101 PMU init succeeded, using AXP2101 PMU

It looks good so far.
I have the "charging battery logo" displayed on the OLED when plugged in.

I will do more test when I have more time, but at the moment, with my TBeam v1.2 and AXP2101 it looks good.
Thanks a lot !

mooiweertje, 2024-02-09:
I have this board aswell but I am not able to get it going. Flashing seems to work but display doesn't show anything and nothing on WiFi. Flashing Meshtastic work fine so it's not broken.

mooiweertje, 2024-02-09:
> I have this board aswell but I am not able to get it going. Flashing seems to work but display doesn't show anything and nothing on WiFi. Flashing Meshtastic work fine so it's not broken. I tried al the v5.5.2 firmwares but no succes.

mooiweertje, 2024-02-10:
Ah, I found the 5.3.5 firmware in jccricket his repo and it seems to work. All that needs to be done is a pull request to get it in this repo? I can do that if you like from my repo.