gereic/GXAirCom #139: CustomGPS config

rvt, 2023-07-11:
Hey all,

I made a PR here : https://github.com/gereic/GXAirCom/pull/133 but since there was no comment or question on it I figured see if I could reach out here and hopefully we can have a discussion/comments and questions going on because I believe this will benefit the flying community.

Basically it allows you to configure and query some of the parameters within GxAirCom over NMEA. The reason why this is usefull is because then it can be configured in automation using other devices, here is such example:

```
2023/07/11 19:07:05 Received GxAirCom Tracker configuration: PGXCF,1,0,0,0,1,1,1,1,5,2,AAAAAA,Stratux
2023/07/11 19:07:05 GxAirCom tracker configuration ok!
```

In the above case stratux requested the configuration from GXAirCom using `$PGXCF,?*59` on which GxAirCom responded with `PGXCF,1,0,0,0,1,1,1,1,5,2,AAAAAA,Stratux`

The format is as follows:
`$PGXCF,<version>,<Output Serial>,<eMode>,<eOutputVario>,<output Fanet>,<output GPS>,<output FLARM>,<customGPSConfig>,<Aircraft Type (hex)>,<Address Type>,<Address (hex)>, <Pilot Name>`
All the parameters already exists as settings within GzAirCom except: `customGPSConfig` and `Address Type` and `Address (hex)`.

Equally, you  can re-configure GxAirCom with `PGXCF,1,0,0,0,1,1,1,1,5,2,AAAAAA,Stratux`

|  Parameter | Explanation   |
|---|---|
| Address Type | Type of Address 1=ICAO  2=FLARM |
| Address (hex)| Address when Address type set to 1 you would usually set Address this to your mode-s |
| customGPSConfig | Reconfigured the GPS uBlox to provide more information on GPS, enables Galileo, 5Hz update rate, enable SBAS and set dynamic model to 2g and a few other settings|

Note that Address and Address type are fully according to the FLARM specification. See http://www.ediatec.ch/pdf/FLARM%20Data%20Port%20Specification%20v7.00.pdf (See also the flarm manuals, but they all seem to download)
[image]
Also Skydemon and other EFB's will understand these when received (as what GxAirCom does).

## How is this used in Stratux?

here is an example configuration:
In the configuration screen Of stratux, when GxAirCom is detected the user is presented with the following screen:
[image]
The user can now select it's name, use it's mode-s HEX and configure stratux/GxAirCom.

From this point onward when the user starts flying, GxAirCom will send over FLARM the ICAO address. Here is how it's received in my OGN base station:

When GXAirCom set as FLARM:
```FLR123456 [   52/   74sec]   8:2:123456 F*  < 0.2m/s> <19.8dB>, <0.0bit/packet>, <-10.12(0.01)kHz>
193743: [ +5X.89278,  +4.X3627]deg     5m  -1.0m/s   0.7m/s 000.0deg  +0.0deg/s 8m5 00x00m O :01f__-10.11kHz 20.2/34.5dB/0  0e     0.0km 000.0deg -20.9deg```

When GxAirCom set as ICAO:
```ICA123456 [   52/   67sec]   8:1:123456 F*  < 0.0m/s> <53.1dB>, <0.0bit/packet>, < -9.93(0.01)kHz>
050728: [ +5X.89287,  +4.XX664]deg    63m  -4.0m/s   0.0m/s 000.0deg  +0.0deg/s 8m1 00x00m O :01f__ -9.94kHz 53.2/67.5dB/0  0e     0.0km 040.8deg +59.2deg```  
```

note: 8:2 and 8:1 means Airplane type 8 and Address type 1 (Motorised plane==8)

### Why setting ICAO.. and why use GxAirCom?
I think GxAirCom is a wonderful product and open source! It allows us to see each other in the sky, and helps taking preventive measures so we do not get any near incidents or worse any accidents.
A  combination of Stratux + GxAirCom allows us to see FLARM/Fanet, Mode-S, SafeSky, ADSB-Out PAW and a few more, but when using GxAirCom with FLARM it also allows planes equipped with FLARM only to see ous. So see and BE seen.

The reason to set ICAO as the same as mode-s is to ensure you do not see yourself hunting down, but also other planes with not see double targets FLARM + MODE-S will then be see as a double target. By setting GxAirCom as ICAO will prevent the double target.

## What about the default behaviour of GxAirCom?

When sending`$PGXCF,1,3,0,1,0,1,1,0,5,2,,GXAirCom*4A` the default behaviour of GxAirCOm is restored. That is it will take again the FLARM address together with the HwAddress and configure the GPS as default. So you could even use GxAirCom in your plane, and still use it as FLARM only in your glider, paraglider etc.

The end goal is to be able to have more aircraft to be seen in the sky in area's where FLARM is more dominant than for OGN or ADSB(out).

Hope thus helps understanding the PR and what we are trying to do!

rvt, 2023-08-04:
this was merged

closed 2023-08-04 by rvt

luar123, 2024-03-16:
Hi @rvt
Thanks for the explanation. Maybe add this to the wiki?
Some questions regarding the implementation:
1. Can the address be overwritten even if address type is set to 2?
2. What happens if address type is set to 1 and Fanet TX is active? Will it use the ICAO address also for fanet packages?
3. The address type in flarm out PFLAA message is broken, see #162 

The fanet protocol does not support an address type and therefore the fanet ID should always be one defined in the protocol, usually the gxaircom address (08xxxx)
However there seems to be the posibility to send the ICAO address in a HW Info message (type A) which is currently not implemented in gxaircom.
See https://github.com/3s1d/fanet-stm32/blob/master/Src/fanet/radio/protocol.txt