gereic/GXAirCom #141: T-Beam S3 Core build

VirusPilot, 2023-07-15:
I have successfully compiled and flashed for the T-Beam S3 but same problem as with the prior T-Beam: when connecting with WiFi and trying to load the web page, the device resets. I could resolve this with the T-Beam by flashing the upstream binaries and then saving the settings on the device but here I can't flash the upstream binaries for the T-Beam S3.

gereic, 2023-07-15:
Have you flashed also the spiffs (filesystem) ?
If not, that is the probelm.

VirusPilot, 2023-07-15:
Here is my flash command: `esptool --baud 921600 --chip esp32s3 -p /dev/cu.usbmodem14201  write_flash 0x1000 bootloader.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 firmware_v5.3.5_T-Beam-S3Core.bin 0x3d0000 spiffs_v5.3.5.bin`

VirusPilot, 2023-07-15:
Is the spiffs partition to be flashed at 0x3d0000 or 0x7d0000?

rvt, 2023-07-15:
I took my info from : https://github.com/gereic/GXAirCom/blob/master/8mb_spiffs.csv

VirusPilot, 2023-07-15:
I now separately flashed spiffs @ 0x7d0000 after the poi flash, now working.

VirusPilot, 2023-07-15:
Last observation today, now that it is running: I can't redirect the GPS output to serial

gereic, 2023-07-15:
What do you mean by serial ? Do you mean the USB-C port or the serial on the pins ?
It sends the GPS to the serial of the pins not to USB-C.

VirusPilot, 2023-07-15:
With serial I was expecting to the USB-C port, similar to older T-Beams.

gereic, 2023-07-15:
You can change the behavor in the platform.ini by uncomment the line
;-DARDUINO_USB_CDC_ON_BOOT=1 ;for output over USB-C

VirusPilot, 2023-07-15:
Will do and report, probably not before tomorrow.

VirusPilot, 2023-07-16:
All fine now. Still would like to know whether the spiffs partition has to be flashed at 0x3d0000 or 0x7d0000 on the ESP32 and the ESP32S3?

gereic, 2023-07-16:
On the 8mb_spiffs.csv (which is used for the S3Core-Board, 0x7d0000
On the min_spiffs.csv (which is used for the other ESP32, 0x3d0000

VirusPilot, 2023-07-16:
no issues any more, thanks for the quick response!

hogmoff, 2023-07-19:
I tried to flash the T-Beam S3 SoftRF with the firmware v5.3.5 from 
https://github.com/gereic/GXAirCom/tree/master/bin
with settings from 
https://github.com/gereic/GXAirCom/blob/master/bin/downloadsettings.JPG
but nothing happens after successful download and reset. Before flash, i've clicked erase.

I tried to change the address from 0x3d0000 to 0x7d0000, too. Same behavior.

What i'm making wrong?

rvt, 2023-07-19:
I don't believe this is in master yet.
If you want to try one of the pull requests, then you have tio build it yourself. Visual Studio Code + PlatformIO is your friend, it's fairly easy if you are a bit technical.
Then upload to your tbeam.

this is what I do for a default T-Beam: 
```
pio run -t upload -e psRam
pio run -t uploadfs -e psRam
```

uploadsfs you usually only have to do it the first time...

VirusPilot, 2023-07-19:
I believe I did the following steps:
- compile/upload the latest sources using PlatformIO (selecting env:T-Beam-S3-Core)
- manually flashing spiffs: `esptool --baud 921600 --chip esp32s3 -p /dev/<your T-Beam> write_flash 0x7d0000 spiffs_v5.3.5.bin`

hogmoff, 2023-07-19:
Thanks for your helpful replies! I'll try it

rvt, 2023-07-20:
> I believe I did the following steps:
> 
> * compile/upload the latest sources using PlatformIO (selecting env:T-Beam-S3-Core)
> * manually flashing spiffs: `esptool --baud 921600 --chip esp32s3 -p /dev/<your T-Beam> write_flash 0x7d0000 spiffs_v5.3.5.bin`

If it requires something special, it might justify creating a new environment for it?

hogmoff, 2023-07-23:
> I believe I did the following steps:
> 
>     * compile/upload the latest sources using PlatformIO (selecting env:T-Beam-S3-Core)
> 
>     * manually flashing spiffs: `esptool --baud 921600 --chip esp32s3 -p /dev/<your T-Beam> write_flash 0x7d0000 spiffs_v5.3.5.bin`

With this settings it's now working. I was a little confused about no blinkings leds, but in PlatformIO with Monitor i saw the working T-Beam. Thanks for your help!

VirusPilot, 2023-08-14:
Seems to be resolved.