gereic/GXAirCom #142: Migration to SparkFun u-blox Arduino GNSS Library v2.2.24

VirusPilot, 2023-07-16:
Some of us are now seriously working on the u-blox configuration code, therefore and particularly for the M10S on the T-Beam S3 Core board it would be beneficial if the SparkFun u-blox library could be updated to the latest version 2.2.24 as it offers full support for generation 8, 9 and 10 u-blox GNSS modules, see here:

[https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library#compatibility](url)

Migration is pretty simple, see here:

[https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library#migrating-to-v20](url)

I am already successfully using this library in my fork.

gereic, 2023-07-20:
Added in latest version

closed 2023-07-20 by gereic

rvt, 2023-07-20:
Sweet, I will modify my code to use this lib aswell