gereic/GXAirCom #145: Support for BMP280

Peuqui, 2023-08-05:
Hi there,
and thanks a lot for your effort!

As I flashed the newest firmware von my TTGO with 1262 ( and Stratux setup V. 29), I noticed, that the BMP280 isn't working as with the SoftRF or OGN Tracker.

Will this being fixed in the future or shall I resolder the BMP280 from the TTGO to the Raspi?

Kind regards,
Peuqui

rvt, 2023-08-05:
top of my head... pins of what OGN is using for the BMP280 is not the same as what GxAirCom is using.

What pins are you currently using for the BPM280?

Peuqui, 2023-08-06:
Thanks for reply!

> If you have soldered the BMP on your t-beam then just leave it and get a new BPM280 and add it to your Raspi. GxAirCom won't use/see it anyways. I found out the hard way this was easer :)
>

I desoldered one from a broken T-Beam: wasn't funny, so I go your way.
But when I revert to OGN on the T-Beam, then the two BMPs may interfere with each other, isn't it?

rvt, 2023-08-06:
Great question, I don't know well enough about OGN and understand what it does when two BMP's are connected.
Nowdays I always use a connector to connect the stuff so it's much easer to disassemble.

Peuqui, 2023-08-06:
Yep, thouht about that too. But when I use the Stratux in the Gyrocopter, I fear, that the vibrations disconnect connectors while not soldered.

b3nn0, 2023-08-06:
@rvt If you deliver the pressure altitude as $PGRMZ, it should be accepted by Stratux just fine, similar to SoftRF and OGN Tracker.
If there are two BMPs in the Stratux, one on the T-Beam and one on the PI, Stratux will only use the one on the PI and ignore $PGRMZ sentences.

Peuqui, 2023-08-06:
`I do have a branch for STratux sitting on my computer that will also support getting baro from GxAirCom to Stratux but opted to not adding that in this version to keep the changes small. If there is traction I can send in a MR to Stratux to have that supported (GxAirCom would still need pin 13/14 though)`

It seems a better way to collect the baro data from the BaroSensor connected to the T-beam, rather than from baro Sensor connected on to the raspi. So the data comes from one single device and the T-Beam is able to transmit the baro data too. Which is not possible, when the baro Sensor sits on the Raspi, isn't it?

b3nn0, 2023-08-06:
> It seems a better way to collect the baro data from the BaroSensor connected to the T-beam, rather than from baro Sensor connected on to the raspi. So the data comes from one single device and the [...]

In theory, yes. In practice, it doesn't matter. Neither Flarm, nor PAW or ADS-L work with pressure altitudes. Only OGNTP and I think FANET have it in an additional, optional field. If it's not there, GPS altitude is used for comparing. IIRC, Stratux even ignores that field when received, so do most others.

source: my bad memory.. might be wrong. But what's not wrong is, that it is, in my opinion, absolutely unimportant.

Peuqui, 2023-08-06:
As I read correctly, it does have impact if altitude is from baropressure or GPS. The flight altitude depends on pressure altitude...

b3nn0, 2023-08-06:
No. Stratux knows what it does. And if it receives an altitude, and knows its source (Baro or GPS), it will do the right thing. And for traffic warnings, it will compare received GPS altitudes to its own GPS altitude, while for received Baro altitudes, it compares those. As long as Stratux knows both altitudes, or can at least get a good estimation, you will get proper traffic warnings.
The only advantage of "Mounting the BMP to the T-Beam" over "Mounting BMP to the PI" is, that the pressure altitude will also be transmitted via OGNTP or FANET, in addition to the GPS altitude. But that is entirely optional in those protocols. If the receiver makes use of this additonal information is up to him. Stratux doesn't do anything with it if it receives a target transmitting both altitudes. The reason is some buggy old firmware that did transmit wrong barometric altitudes. Therefore, Stratux just does the conversion itself:
https://github.com/b3nn0/stratux/blob/7903eb48781a53f5a70d9d90e4965386d3eeb002/main/ogn.go#L272

Conclusions:
- Baro altitude TX is pretty much irrelevant anyway
- It doesn't matter if Stratux uses either Baro. Internal i2c baro has the advantage of delivering a faster update rate and Stratux has better control over what happens, that's why we use that.
- If you still want to force it to use the T-Beam one, feel free to disable Baro i2c in the settings. T-Beam pressure altitude will still be used.

Peuqui, 2023-08-06:
Thanks for that comprehensive explanation!