gereic/GXAirCom #146: 5.4.0 on Heltec V1

sbishop61, 2023-08-24:
V5.3.4 works on my Heltec V1.
When I flash v.5.4.0 I can connect to the wifi but when I try to load the page  (192.168.4.1) it crashes:

`[I][main.cpp:1359] WiFiEvent(): a station connected to ESP32 soft-AP
[I][main.cpp:1365] WiFiEvent(): soft-AP assign an IP to a connected station
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d7c64  PS      : 0x00060b30  A0      : 0x800d7e71  A1      : 0x3fff2b60
A2      : 0x3fff36d4  A3      : 0x00000000  A4      : 0x3fff2ba4  A5      : 0x3ffcd6ac
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800d7c61  A9      : 0x3fff2b40
A10     : 0x3fff2b6c  A11     : 0x3f407925  A12     : 0x6dd24535  A13     : 0x3ffcd6ac
A14     : 0x00000000  A15     : 0x3fff2b6c  SAR     : 0x00000017  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff9

ELF file SHA256: 0000000000000000

Backtrace: 0x400d7c64:0x3fff2b60 0x400d7e6e:0x3fff2ba0 0x40126fc9:0x3fff2be0 0x40124d95:0x3fff2c20 0x40124                            e5d:0x3fff2c60 0x401250b1:0x3fff2cb0 0x401221f5:0x3fff2cd0 0x40122289:0x3fff2d10 0x401229b9:0x3fff2d30 0x4                            0092032:0x3fff2d60

Rebooting...`

gereic, 2023-08-25:
Can you try to erase flash and reflash the firmware and spiffs ?

sbishop61, 2023-08-25:
I did a full erase and flash with all files.
I will try again

sbishop61, 2023-08-26:
I have tried a second board. Full erase and flash. Same behaviour.

`ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[I][FanetLora.cpp:85] setGPS(): disable GPS PPS
[I][main.cpp:1794] setup(): SDK-Version=v3.3.5-1-g85c43024c
[I][main.cpp:1795] setup(): CPU-Speed=240MHz
[I][main.cpp:1796] setup(): XTAL-Freq=26MHz
[I][main.cpp:1797] setup(): APB-Freq=80000000Hz
[I][main.cpp:1798] setup(): Total heap: 252584
[I][main.cpp:1799] setup(): Free heap: 225444
[I][main.cpp:724] printChipInfo(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE,
[I][main.cpp:725] printChipInfo(): silicon revision 0,
[I][main.cpp:727] printChipInfo(): 4MB external flash
[I][main.cpp:1816] setup(): compiled at Aug 24 2023 20:41:47
[I][main.cpp:1817] setup(): current free heap: 195276, minimum ever free heap: 190796
[I][main.cpp:1568] print_wakeup_reason(): wakeup undefined --> possible by reset
[I][main.cpp:1828] setup(): SPIFFS total=173441 used=0 free=173441
[I][fileOps.cpp:6] load_configFile(): LOAD CONFIG FILE
[E][Preferences.cpp:492] getBytesLength(): nvs_get_blob len fail: BATOFFS NOT_FOUND
[I][main.cpp:1838] setup(): set CPU-Speed to 240MHz
[I][main.cpp:1454] printSettings(): **** SETTINGS v5.4.0 build:Aug 24 2023 20:41:47 ******
[I][main.cpp:1455] printSettings(): Access-point password=12345678
[I][main.cpp:1456] printSettings(): Board-Type=1
[I][main.cpp:1457] printSettings(): Display-Type=1
[I][main.cpp:1458] printSettings(): Display-Rotation=0
[I][main.cpp:1459] printSettings(): Mode=0
[I][main.cpp:1460] printSettings(): Fanet-Mode=0
[I][main.cpp:1461] printSettings(): Fanet-Pin=0
[I][main.cpp:1462] printSettings(): external power switch=0
[I][main.cpp:1465] printSettings(): Serial-output=0
[I][main.cpp:1466] printSettings(): OUTPUT Vario=1
[I][main.cpp:1467] printSettings(): OUTPUT FLARM=1
[I][main.cpp:1468] printSettings(): OUTPUT GPS=1
[I][main.cpp:1469] printSettings(): OUTPUT FANET=1
[I][main.cpp:1470] printSettings(): Device ID=
[I][main.cpp:1471] printSettings(): Device Address Type=2
[I][main.cpp:1473] printSettings(): WIFI connect=0
[I][main.cpp:1474] printSettings(): WIFI SSID=
[I][main.cpp:1475] printSettings(): WIFI PW=
[I][main.cpp:1476] printSettings(): Aircraft=PARA_GLIDER
[I][main.cpp:1477] printSettings(): Pilotname=
[I][main.cpp:1478] printSettings(): Wifi-down-time=180
[I][main.cpp:1479] printSettings(): Output-Mode=3
[I][main.cpp:1480] printSettings(): UDP_SERVER=192.168.4.2
[I][main.cpp:1481] printSettings(): UDP_PORT=10110
[I][main.cpp:1482] printSettings(): UDP_SERVER=192.168.4.2
[I][main.cpp:1483] printSettings(): UDP_PORT=10110
[I][main.cpp:1486] printSettings(): GS LAT=0.000000
[I][main.cpp:1487] printSettings(): GS LON=0.000000
[I][main.cpp:1488] printSettings(): GS ALT=0.00
[I][main.cpp:1489] printSettings(): GS SCREEN OPTION=0
[I][main.cpp:1490] printSettings(): GS POWERSAFE=0
[I][main.cpp:1492] printSettings(): BattVoltOffs=0.00
[I][main.cpp:1493] printSettings(): minBattPercent=20
[I][main.cpp:1494] printSettings(): restartBattPercent=20
[I][main.cpp:1496] printSettings(): AirWhere-Livetracking=0
[I][main.cpp:1497] printSettings(): OGN-Livetracking=0
[I][main.cpp:1498] printSettings(): Traccar-Livetracking=0
[I][main.cpp:1499] printSettings(): Traccar-Address=
[I][main.cpp:1500] printSettings(): RF-Mode=11
[... 49 lines cut ...]
[I][main.cpp:1400] setupWifi(): Setting soft-AP configuration ...
[I][main.cpp:1402] setupWifi(): Ready
[I][main.cpp:1314] WiFiEvent(): station stop
[I][main.cpp:1314] WiFiEvent(): station stop
[I][main.cpp:1430] setupWifi(): my APIP=192.168.4.1
[I][main.cpp:1359] WiFiEvent(): a station connected to ESP32 soft-AP
[I][main.cpp:1365] WiFiEvent(): soft-AP assign an IP to a connected station
[I][main.cpp:1359] WiFiEvent(): a station connected to ESP32 soft-AP
dhcps: send_nak>>udp_sendto result 0
[I][main.cpp:1365] WiFiEvent(): soft-AP assign an IP to a connected station
[I][main.cpp:1365] WiFiEvent(): soft-AP assign an IP to a connected station
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d7c64  PS      : 0x00060630  A0      : 0x800d7e71  A1      : 0x3fff3a10
A2      : 0x3fff531c  A3      : 0x00000000  A4      : 0x3fff3a54  A5      : 0x3ffcd6ac
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800d7c61  A9      : 0x3fff39f0
A10     : 0x3fff3a1c  A11     : 0x3f407925  A12     : 0x49996e10  A13     : 0x3ffcd6ac
A14     : 0x00000000  A15     : 0x3fff3a1c  SAR     : 0x00000017  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff9

ELF file SHA256: 0000000000000000

Backtrace: 0x400d7c64:0x3fff3a10 0x400d7e6e:0x3fff3a50 0x40126fc9:0x3fff3a90 0x40124d95:0x3fff3ad0 0x40124e5d:0x3fff3b10 0x401250b1:0x3fff3b60 0x401221f5:0x3fff3b80 0x40122289:0x3fff3bc0 0x401229b9:0x3fff3be0 0x40092032:0x3fff3c10

Rebooting...`

VirusPilot, 2023-08-26:
Did you try flashing the firmware and spiffs in two separate steps? In my case this helped ...

sbishop61, 2023-08-28:
I have tried that and it works. Any idea why?

VirusPilot, 2023-08-28:
no idea, @gereic?

closed 2023-09-21 by gereic