gereic/GXAirCom #148: [Feature Request] Flexible WiFi settings.

bsiege, 2023-08-30:
Separate APMode and Station mode. 

Benefit

As Ground Station APMode is not needed. ESP is accessible over the connected network. No need to broadcast unneeded SSID

gereic, 2023-09-21:
Hi,
I added this feature in the settings.
Please try and give feedback.

bsiege, 2023-09-22:
Looks promising, annoying beacon is gone ;-) Thanks!!

P.S. Realized that the bandwith in AP mode is 40Mhz. Makes no sense for this application. You could go down to 20 without performance issue, but save a remarkable amount of battery power.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv416wifi_bandwidth_t
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv422esp_wifi_set_bandwidth16wifi_interface_t16wifi_bandwidth_t

And for even more airtime for Pilots with wifi setup and distance of a few centimeters between devices reducing power would tackle another percents..
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_max_tx_power6int8_t