gereic/GXAirCom #149: UF2 file for ESP32 S3

Toogo, 2023-08-31:
Can you publish a UF2 file to upload to an ESP32 Supreme.

Frank

VirusPilot, 2023-08-31:
You can do it on your own, you just need to download the script from here:

https://github.com/microsoft/uf2/tree/master/utils

The command is the following:

uf2conv.py infile.bin -c -b 0x00 -f ESP32S3 -o outfile.uf2

But I don't really see a good reason to stay with uf2 only because the modules are shipped with SoftRF in a uf2 configuration. Simply put your Supreme board in Espressif Service Mode (press and keep holding BOOT, press and release RESET, thereafter release BOOT) and then flash GXAirCom directly - this only needs to be done once.

Toogo, 2023-08-31:
Ok, but what are te settings and Mem address, does it need to be on 80Mhz, QiO?
All these files?

boot_app0.bin
bootloader_dio_40m.bin
firmware_v5.4.0_T-Beam-S3Core.bin
partitions.bin
spiffs_v5.4.0.bin

VirusPilot, 2023-08-31:
You need to flash in two steps:
1. esptool --baud 921600 --chip esp32s3 -p /dev/your_serial_port write_flash 0x1000 bootloader.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 firmware_v5.4.0_T-Beam-S3Core.bin
2. esptool --baud 921600 --chip esp32s3 -p /dev/your_serial_port write_flash 0x7d0000 spiffs_v5.4.0.bin

Step 1 can also be performed inside PlatformIO

Step 2 is only required once, further firmware updates only require step 1 (unless the spiffs layout is changed)

VirusPilot, 2023-08-31:
Did you put the Supreme in service mode? I would simply try flashing with PlatformIO first ...

Not sure about the bootloader file ...

Toogo, 2023-09-01:
I did put it in Service mode, I will fiddle more :-), Thank you.

slash-bit, 2024-02-28:
Hi , did that get resolved? I am also flashing Supreme ESP32 S3 in Service mode , using Espresif Flash Download tool. I  tried with 2 step approach.
After restarting I am also getting COM4 on and off repeatedly.