gereic/GXAirCom #155: E-Ink Display on T-Beam V1.2

ralphem, 2023-11-05:
Hello, i have an strange effect with e-ink display. the 2,9 eInk(V1)  is function only wiht software parameter V2 of E-Ink. only the wire must very shortly cut . It´s dont need remove any resistors.
The display  2,9 inch  V2 dosn´t function ( also not with short wire). 
I had change the display to 2,13 inch and function with software parameter (E-Ink V2) directly. But some lines was trapped.
This is reproducibly.
Have anyone  experience for that connect Wave Share E-Ink 2,9 Inch V2 ?
Another question. The BUZZER  must connect to PIN 25 on. I had tested 2 different type ( 2 wire and 3 wire ´plus/minus/signal´) , but no sound . With oszilloscope i cannot found  any signal on pin 25.  The display and webfrontend switched for low, mid and high signal.
Was an change in the firmware , i had tested V5.3.x. / 5.4.0. and 5.4.1 ?
Have anyone  experience therefor?

Best regards

mikechristiansen, 2023-11-06:
Hi Ralphem,
Ive had some issues recently with the 2.9" V2 display. See this faccbook post. https://www.facebook.com/groups/1130097100726170/permalink/1853962685006271/?mibextid=uJjRxr 
Essentially I was having an issue with any reboot the screen not working. I ended up swapping pin 25 with pin 0 so that my e-ink reset is now 25 and the buzzer is now 0. I compiled the code with these changes, changed my wiring and now Its working fine. 
I believe there was a change made in code at version 3 swapping these. 

I think you are experiencing the same issue I had as your also having buzzer issues. 

Hope this helps!

ralphem, 2023-11-08:
Hi mikechristiansen, many thanks for your post. i have read your facebook tweet. it´s verry interesting.
My english isn´t very good.
i´am not programmer, and i cold not found an .ino file (for Arduino IDE - GXAircom master) for compilation in new config (change port for buzzer). I have only probe compilation the softrf. these i can found here in github.
in my config is port 0 open. i found the port config for 2,9 E-inc. also  how your set.
RST - 0 ( i have changed to VCC)
DC-32
CS-15
CLK-4 
DIN-2
Busy-33.
in another tweet i found the RST Change for VCC
normal i use the BIN files and flash the T-Beam with ´flash-dowload tool´, it´s easyer for me. When do you have the software (GXAircom _master ) for compilition itself, i would like take this. (for Arduino IDE?)

one question i have: do you have remove 2 resistores in the right side by USB connector?

Best regards

mikechristiansen, 2023-11-09:
Ralphem,
I did remove the resistors - did not try the screen prior. 
You can compile the firmware using visual studio and platformIO - ive just learnt this program myself! 

These are the changes I made. 

 //V3.0.0 changed from PIN 0 to PIN 25
  // PinBuzzer = 25; - Changed back to 0 for E-Ink.
  PinBuzzer = 0;

  //E-Ink
    PinEink_Busy   =  39;
  //PinEink_Rst    =  0; - changed to 25
    PinEink_Rst    =  25;
    PinEink_Dc     =  15;
    PinEink_Cs     =  13;
    PinEink_Clk    =  14;
    PinEink_Din    =  2;

Hope this helps!

ralphem, 2023-11-09:
Hi mikechristiansen, i hade tested an T-Beam V1.0, and what can i say, is function without change. now i will search in the next time what is the issues in this case with V1.2 T-Beam. 
Many thanks for your help.

closed 2023-11-11 by ralphem