gereic/GXAirCom #157: problem TTGO TBEAM T22 V1 / GXAIR COM on STRATUX / New antenna GPS 28db / Shudown power

francois31370, 2023-12-20:
I installed a 28db gain GPS antenna on my TTGO Tbeam module connected via USB to a Raspberry Pi for Stratux. After a few minutes of use after receiving a position, the TTGO shuts down. I have to reboot the Stratux so that GXAirCom restarts and then shuts down again after a few minutes. I tried with 4 different TTGOs and different antennas and it's the same. I use the latest version of Stratux EU and GXAirCom. I think GXAIRCOM detects too high a current demand and then stops. I should point out that the TTGO no longer has a battery and is powered via USB by the Raspberry Pi 3B+

rvt, 2023-12-20:
How may SDR dongles do you have attached? I noticed for myself that when I used a RPI4 and with two SDR dongles that power supply is critical.

If you do have a SDR, can you try to disconnect them just to see if you still see these reboots?

francois31370, 2023-12-20:
It's 2 SDR nooelec Nano 2+
Ok i will try to disconnect one.
TY

francois31370, 2023-12-21:
It was the current to low from 4 USB. I cut the 5v wire from an USB cable and plug it direcly to raspberry Pi for power the TTGO. Now it's ok. I will try with Raspberry 4b because the maxim current from the 4 USB is higher.

rvt, 2023-12-21:
You can get power deliver (PD) modules that ensures that powerbank with USB-C, that suppoers PD you can get the appropriate current. Than feed that directly on the HAT pins (if you are brave enough). I have used that with good success. That also then allows you to use a proper power plug on the stratux instead of the flimsy USB-C/Micro-USB plugs..
You need PD, else the powerbank will limit it to 0.5 or 1A..