gereic/GXAirCom #164: Recent Flarm protocol update considered ?

Fluchbaer, 2024-04-01:
Following some information [https://lu-glidz.blogspot.com/2024/03/das-livetracking-fiasko.html](url) the FLARM protocol has been recently changed which influenced the OGN tracking already.
Is the GXAircom software affected ? I can't test it due to the lack of an other FLARM device.
The latest listed firmware is 5.5.2, while the firmware history description in version.txt ends with the 5.5.1.

rvt, 2024-04-01:
Unfortunately both SoftRF, GxAirCom and my own Firmware is affected and that's a sad to for aviation safety.

Flarm Changed the protocol, we would never know the reasons except we can only guess protecting their business.

gereic, 2024-04-02:
At the moment, I will not support the new Flarm-Protocol.
There is a new Project starting
https://www.project-ocap.net/
I'm already in contact with them and as far as it is finished, I will support it.

closed 2024-04-02 by gereic

rvt, 2024-04-02:
@gereic I hope it's based on ADS-L specifications, otherwise it's yet a other protocol others need to support, Will ask them to be sure, other whise this applies: https://xkcd.com/927/

gereic, 2024-04-05:
@rvt 
It seems, that Flarm itself doesn't want to use ADS-L
[Wiki ADS-L](https://de.wikipedia.org/wiki/Automatic_Dependent_Surveillance)

ADS-L unterstützt zwei physikalische Systeme; das [M-Band](https://de.wikipedia.org/wiki/Frequenzband), das im Spektrum von 868–868,6 MHz arbeitet und das O-Band, das im Spektrum von 869,4–869,65 MHz arbeitet.[[10]](https://de.wikipedia.org/wiki/Automatic_Dependent_Surveillance#cite_note-easa-ads-l-spec-10) Dies verhindert eine direkte Nutzung als Kollisionswarnsystem, da jedes Gerät auf beiden Frequenzbänder hören müsste, ob ein anderes Flugobjekt in der Nähe ist. Gemäß der Firma [Flarm](https://de.wikipedia.org/wiki/Flarm) eignet sich ADS-L daher nur als Tracking-System und nicht als Antikollisions-System, da wichtige Teile für eine effektive Kollisionsverhinderung fehlen

rvt, 2024-04-05:
I think I did mean that I open that https://www.project-ocap.net/ will use ADS-L.. Flarm just made aviation again a little bit less safe again :(

gereic, 2024-04-05:
Project-ocap does actually the following:

- Check, what is stand of the art
- Check frequencys to analyze what is necessary to create a protocol for collision-awarness.

After the tests are finished, a protocol will be specified.
If it will bei ADS-L or not, depends on the tests before.