gereic/GXAirCom #165: Xctrack + GXAir

AlexSA6002, 2024-04-15:
Hello. 
the device works fine
But in my xctrack other devices are not displayed as well as yours
[image]

your xctrack 

[image]

PIDIENNE, 2025-02-19:
Did you have any progress since?
Is it possible to have a sound notification on XCTrack (connected to GXAirCom via BT) when another aircraft is on a convergent path?

gereic, 2025-02-20:
Can you describe the problem better ?
Which device have you connected to xctrack ?
Which device do you see and which one not.
In which country are you ?

AlexSA6002, 2025-02-20:
I’m from Russia. Connected device with SoftRF Fanet or device with GXair firmware via BT. On my screen you can see an incomprehensible set of numbers under the indicator of another plane. On yours you can see the Fanet ID and altitude. That's the problem. Why is it like this for me and will it be fixed?

luar123, 2025-03-08:
Could be related to #159.

gereic, 2025-03-09:
I have implmented the bugfix in the lastest version.
Can you try it please ?

AlexSA6002, 2025-03-10:
great i'm trying this week