gereic/GXAirCom #167: No GPS Signal

FreezeManny, 2024-05-23:
Hi,
i have a T-Beam V1.2 (The V1.1 with the AXP2101) and cannot get a GPS Fix.
When using SoftRF, I recieve 6 Satelites. 
Also no LEDs on the T-Beam are lighting up, but it does create a web Interface. (No OLED connected)
Greetings

gereic, 2024-05-26:
Hi,
First, you can try to press the configure GPS button in settings-page.
Second, check, if you get $GPRMA or $GPPPA sentence.
When you set output to serial, you can check that easily with an serial terminal.

parapenT1sta, 2024-06-21:
> Hi, i have a T-Beam V1.2 (The V1.1 with the AXP2101) and cannot get a GPS Fix. When using SoftRF, I recieve 6 Satelites. Also no LEDs on the T-Beam are lighting up, but it does create a web Interface. (No OLED connected) Greetings

In my case the latest version I could get working properly was 5.6 0 with my T-BEAM v0.7. After this version no matter how long I wait for GPS Fix or change settings I can make it working.

karhun2023, 2024-07-09:
Hello everyone, 
I have the identical issue. For me, reverting to an earlier version does not help. The power supply to the Ublox is there. However, only $GPRMC and $GPGGA sentences are coming through.

The following comes from version 5.6.0.

One Error...  [E][main.cpp:1128] setupPMU(): failed to found any PMU (AXP192, AXP2101)

and some Data.
 $PFLAU,0,1,0,1,0,0,0,0,0*52
$LK8EX1,95971,455.54,21,30.0,1000,*24
$LK8EX1,95972,455.56,20,30.0,1000,*24
$GPRMC,,V,,,,,,,,,,N*53
$GPGGA,,,,,,0,00,99.99,,,,,,*48
$LK8EX1,95972,455.56,17,30.0,1000,*20
$LK8EX1,95972,455.56,13,30.0,1000,*24
$LK8EX1,95972,455.56,11,30.0,1000,*26
$LK8EX1,95972,455.55,8,30.0,1000,*1D
$LK8EX1,95972,455.54,6,30.0,1000,*12
$LK8EX1,95972,455.54,4,30.0,1000,*10
$LK8EX1,95970,455.56,7,30.0,1000,*13
$LK8EX1,95969,455.63,17,30.0,1000,*2C
$PFLAU,0,1,0,1,0,0,0,0,0*52
$LK8EX1,95970,455.64,16,30.0,1000,*22
$LK8EX1,95971,455.66,16,30.0,1000,*21
$GPRMC,,V,,,,,,,,,,N*53
$GPGGA,,,,,,0,00,99.99,,,,,,*48

No new GPS data even outdoors after 30 minutes.

karhun2023, 2024-07-17:
Sorry... Damn stupid coincidence. With the update to version 6.0, I soldered the battery to the pad under the GPS module. Apparently, the module on the other side came loose. I resoldered all the contacts of the GPS module, and bingo, everything is working as it should.

Bender-bj, 2024-07-26:
> Hi, i have a T-Beam V1.2 (The V1.1 with the AXP2101) and cannot get a GPS Fix. When using SoftRF, I recieve 6 Satelites. Also no LEDs on the T-Beam are lighting up, but it does create a web Interface. (No OLED connected) Greetings

Hello,
I have a T-Beam v1.2 with GXAirCom v6.0.0
With the OLED screen not connected it was impossible to have a GPS fix
Yesterday i soldered and activated the OLED screen in the settings and it works fine.

parapenT1sta, 2024-07-28:
In my case I flashed the latest 6.0.1 and let the whole night trying to get Satellites, It took more than usual but it's working at least with Beam v0.7

mavicquax, 2024-09-29:
Hello gereich,
did you ment $GPGGA sentence instead of $GPPPA in your answer on may 26th?
My problem: 
T-beam 1.1 with replaced GPS modul works fine with some arduino code.
On putty I only get the sentences $GPFLAU, after GPS FIX!
your "GPS configure" in your GXAircom settings shows only an error message.

FreezeManny, 2024-09-29:
Sadly I cannot help you anymore.
My T-Beam seams to be faulty, which is why I replaced it with a regular U-Blox 8 based GPS.
I never got a reliable GPS Signal, sometimes it worked. Sometimes it just didn’t. At one point, after a couple of flights without it, I gave in and replaced the thing.

mavicquax, 2024-09-30:
...additional info about my comment from yesterday!

Flashing the OGN-Tracker sw provided in the newest stratux version 
[Stratux Europe Edition 1.6r1-eu032](https://github.com/b3nn0/stratux/releases/tag/v1.6r1-eu032)
works perfect. (Installation procedure is explained in the stratux wiki)
Great work!!