gereic/GXAirCom #169: Unable to see any SoftRF devices

cyberorg, 2024-09-22:
Hi

I have tried T-Echo with SoftRF firmware, and also tried Legacy, OGNTP, FANET protocols and also tried different frequencies like EU, IN etc., the devices do not show up on "received aircrafts". The SoftRF devices see each other.

I have skytraxx that shows up fine.

I am using T-Beam as Ground Station.

[image]

slash-bit, 2024-09-22:
SoftRF shows as received aircraft in FANET mode.
Legacy I believe no longer supported by GXAIRCOM.
I used SoftRF version 1.51 on T-Echo device.
And GXAIRCOM running on Heltec. The GXAircomm is controlled by XCGuide app on Android.

gx.png (view on web)<https://github.com/user-attachments/assets/ea1574c9-5c61-439f-a198-26840604d48d>

cyberorg, 2024-09-22:
Hi, thanks for the response.

See https://github.com/lyusupov/SoftRF/issues/171

Is there any particular configuration I need to use for them to show up on GXAircom? 

I'm using this: $PSRFC,1,0,5,1,7,1,0,2,2,1,0,1,1,5,0,0,0,0,0*4F

SoftRF <-> SoftRF works
GXAircom <-> Skytraxx works
SoftRF <-> Skytraxx doesn't work
SoftRF <-> GXAircom doesn't work

slash-bit, 2024-09-22:
That's a strange one.
The screenshots of settings of  SoftRF look correct,  the same as mine.
In SoftRF configuration I use Region = EU
Also my T-Echo hardware is 868MHz version.
I think they also sell 915Mhz which will not work with EU region.
Please check you T-Echo hardware version.
What is region on you Skytraxx?

I am not sure where do you put string
$PSRFC,1,0,5,1,7,1,0,2,2,1,0,1,1,5,0,0,0,0,0*4F
Somewhere in GXAIRCOM webpage?

See lyusupov/SoftRF#171<https://github.com/lyusupov/SoftRF/issues/171>

cyberorg, 2024-09-22:
T-Echos used  are 868 version. However what I understand is that just the antenna tuning is different, up close it doesn't matter much.

I'm using web terminal to configure t-echos, that is the string to configure it.

I've used EU frequency also. Do you leave the device on default OGNTP or switch to FANET as I have done?

VirusPilot, 2024-09-22:
The $PSRFC string is usually sent to the SoftRF device only once to configure it. This can be done by connecting e.g. the T-Echo to your PC's serial console. Make sure to select the appropriate baud rate. @cyberorg, if you are brave, you may flash an alternative SoftRF version from @moshe-braner: https://github.com/moshe-braner/SoftRF/tree/master/software/firmware/binaries/nRF52840/SoftRF/MassStorage

VirusPilot, 2024-09-22:
Moshe's fork has a lot of enhancements, e.g. a separate configuration page on the T-Echo where you can change settings without a serial console.

VirusPilot, 2024-09-22:
My SoftRF devices (T-Beam, T-Beam Supreme, T-Echo) are all running fine and are well received by Stratux, OGN stations and other aircraft. Never tried GXAirCom ...

cyberorg, 2024-09-22:
Hi @VirusPilot

That is the only way I am aware of configuring t-echos. Softrf-tool android
app device doesn’t work for me.

I’ve tried all 3 available firmwares: factory, Moshe’s and the latest from
Lyusupov.

None of them show up on GXAircom.

cyberorg, 2024-09-23:
I can finally see SofRF devices, but by using older firmware GXAirCom-v4.12.0!

Thanks to @lyusupov for pointing me to old firmware repository: https://github.com/gereic/GXAirCom/tree/68e2544adf4e6f3a3af4782db2dd74a077dbc0a5/bin

Can the functionality that existed in older versions be brought back?

slash-bit, 2024-09-23:
It works on GXAirComm version 5.6
I will update to 6.1 and report back .

The SoftRF-tool android via Bluetooth stopped working with T-Echo a while ago.
But it works via Serial USB. I just connect T-Echo to my phone by USB-c to USB-C cable directly.

Here is the snapshot of my settings that works with GXAIRCOM in Fanet mode.

slash-bit, 2024-09-24:
Also tested with v6.0 1 (latest) GxAircom  and FANET works.
Two aircrafts received the ID 200 is XCTRACER Maxx2 and 7D is SoftRF.

cyberorg, 2024-09-25:
Can you share your SoftRf version and config details please. Is it the original or Moshe’s? 

On 4.12 fanet works, on 5.5.2 I can get T-Echo only by using Legacy protocol. On 6.0.1 it doesn’t show on any protocol.

This is t-echo with Legacy.
[image]

gereic, 2024-09-25:
Hi, 
As you can see already in the settings, Flarm is not supported in the latest version.
Because Flarm changed there protocol and there is no full documentation about the new protocol.
softrf has implemented the new protocol more ore less.
So I decided to not implement it until it is fully decoded and documented.

You have to set all devices to Fanet and it should work out of the box.

cyberorg, 2024-09-25:
Hi @gereic When using FANET on later versions of GXAirCom the t-echo SoftRF devices do not show up. Only Skytraxx FANET device shows up.

On 4.12 they show up. On 5.5.2 it shows up only when using Legacy, not FANET.

moshe-braner, 2024-09-25:
The new FLARM protocol - like the old one - will never be "fully decoded and documented", they keep it secret.  But enough is known about the new protocol that I would say it is usable.  My version of SoftRF implements all that is known.  Aircraft type, lat, lon, alt, course, speed, turnrate, and more.  For some reason lusupov's version only implements some of the data fields.  (The "factory" version is just an older version of lyusupov's.)

SoftRF is not common, but FLARM is very common in Europe, most gliders have FLARM but not FANET.  So you may want to support the FLARM protocol - and can test it.  OGN ground stations listen to FLARMs and also have been tested to receive my version of SoftRF in "Latest" mode (compatible with the new FLARM protocol).  Also, in tests, FLARMs receive my SoftRF in "Latest" mode, and vice versa.

Note that lyusupov's recent versions in "Legacy" protocol alternate transmitting in the old and new protocols, which may explain why they may be received by older versions of GXAirCom but not the latest version.  In my version you choose either the "Legacy" (old) protocol or the "Latest" (new) protocol for transmissions.  (It receives both when in either protocol, but not if it is set to FANET.)

slash-bit, 2024-09-25:
@cyberorg HI, the SoftRF version I am using is 1.5.1 (original, latest).
The config screenshot is here
[image]
I would really recommend to get SoftRF-tool working via USB and have a look for any difference in config.
I can't think what could be causing the FANET not to work on GxAirCom v6.0.1.
Its quite simple, both devices have to have GPS fix ofcoarse to see each other,

slash-bit, 2024-09-25:
@moshe-braner Hi, thanks for explanations about your version. I will give it a try.
When I used original lyusupov SoftRF version with FLARM it shows on XCTRACER Maxx2 and Ground station (RaspPI 0.3.2)
But it is difficult to say which version of FLARM is received , as you said, lyusupov's version alternating V6 and V7.

Because of problem with XCTRACER Maxx2 can not RX FLARM and GxAIrCom can not RX/TX FLARM. I am having to put third device on my cockpit specifically for FLARM . That device is running on SoftRF. :-)

cyberorg, 2024-09-26:
Finally we’re getting somewhere narrowing down the issue.

Leaving position details to 0 under ground station settings gets SoftRF device to show up. BUT no Skytraxx!

[image]
[image]

Filling in the location shows Skytraxx but no SoftRF!
[image]
[image]

On 4.12 both devices show up when location is filled in.

gereic, 2024-09-26:
GxAircom needs GPS-Position, to check on which frequency to transmit and receive.
Can you post the coutput from the terminal ? (serial console), there you should see the frequency.

gereic, 2024-09-26:
To be honest, your location is untested, so maybe there is something wrong.

cyberorg, 2024-09-26:
Indeed something is wrong, but where.

Skytraxx shows GXAircom(aloft) and 07(softrf, sometimes, either EU or IN).

GXAircom just shows Skytraxx.

gereic, 2024-09-26:
I Think I see the problem.
On the documentation of fanet there is written: 
					.name = "IN866",
					.mac = {.channel = CH_866_200, .dBm = 14, .bw = BW_250},
					.bb = BoundingBox(deg2rad(40), deg2rad(5), deg2rad(89), deg2rad(69))
Which should be your location. (Frequency 866.2Mhz Bandwidth 250kHz
On Softrf:
Europe , Russia , India | 868.2 | 250
so Frequency is 868,2Mhz and Bandwidth 250kHz.

This is the reason, why GxAircom doesn't see SoftRf.
So there is a difference between Skytraxx Frequency and SoftRf.
GxAircom uses the Frequency of Skytraxx.
I can ask somebody from Skytraxx, if the Frequency for your location are right.

moshe-braner, 2024-09-26:
In SofRF, at least in my version which is based on an older version of lyusupov's, in the OGN library, in freqplan.h, I see:
case RF_BAND_IN: { BaseFreq=866000000; ChanSepar=200000; Channels= 1; ...}

gereic, 2024-09-26:
Is this for Fanet-protocol?
If yes and you can change to 866.2Mhz, just try it please.

gereic, 2024-09-26:
In the latest Version of Softrf:
      case RF_PROTOCOL_FANET:
        { ChanSepar=400000; Channels= 1; }
        switch (Plan)
        {
          case RF_BAND_US:
          case RF_BAND_AU:
          case RF_BAND_CN: /* ? */
            BaseFreq   = 920800000;
            Bandwidth  = RF_RX_BANDWIDTH_SS_250KHZ; // BW500
            MaxTxPower = 15;
            break;
          case RF_BAND_IN:
            BaseFreq   = 866200000;
            Bandwidth  = RF_RX_BANDWIDTH_SS_125KHZ; // BW250
            MaxTxPower = 14;
            break;

So, the Frequency should be correct.

slash-bit, 2024-09-26:
@cyberorg  OR instead of changing frequency
In  SoftRF set region to EU
in GxAirComm ground station settings set coordinates to somewhere in Europe

gereic, 2024-09-26:
What happened when you do this ?
Is GxAircom then still visible on the Skytraxx ??

cyberorg, 2024-09-26:
@slash-bit https://github.com/gereic/GXAirCom/issues/169#issuecomment-2375843337 leaving position unset practically does that. Skytraxx won’t show in that case because it is not on EU frequency.

@gereic what does this mean?

```
[I][fmac.cpp:1001] setRegion(): set Lora-Region to IN866
[I][fmac.cpp:1028] setRegion(): zone=1,**freq=868.2**,channels=0 
```

Noticed freq=868.2?

gereic, 2024-09-26:
Forget the last line, because this is the Flarm-frequency and Flarm is not supported anymore.

gereic, 2024-09-26:
So then it is maybe a Problem with Softrf.
Why do you need Softrf?
You can also use GxAirom.

cyberorg, 2024-09-26:
On T-echos?

slash-bit, 2024-09-26:
Yes, looks like frequency is incorrect for IN region in SoftRF. (Y)

GXAirComm on T-Echo would be nice though :-)

moshe-braner, 2024-09-26:
Sorry, I cannot find where in SoftRF (my version, or lyusupov's 1.0 or 1.2) the FANET frequency is set.  Can you tell me which files to look in?  I already did searches in both the SoftRF source and the libraries, not found.

cyberorg, 2024-09-27:
@moshe-braner Here is the latest on frequencies we can use: 865-868. https://thc.nic.in/Central%20Governmental%20Rules/use%20of%20low%20power%20Equipment%20in%20the%20frequency%20band%20865%20to%20868%20MHz%20for%20Short%20Range%20Devices%20Exemption%20from%20Licence%20Rules,2021.pdf

cyberorg, 2024-09-27:
The issue is fixed by changing the IN frequency to 8662000 in SoftRF, thanks @gereic @moshe-braner and everyone who patiently helped debug this.

https://github.com/moshe-braner/SoftRF/pull/19

closed 2024-09-27 by cyberorg

moshe-braner, 2024-09-27:
Not clear whether you mean 866.2 (rather than 866.0) only for LORA or also for FSK (FLARM-compatible protocol).