gereic/GXAirCom #172: T-Beam V1.2 not communicating correct GPS data to Stratux

mpfusion, 2024-10-25:
When using the ogntracker stratux recognises GPS signals correctly, the number of satellites is reported and the GPS button lights up green:

[image]
[image]

However, when I flash to GXAirCom (using the command `install-GxAirCom-Stratux-firmware.sh` from stratux, the output is as follows:

[image]
[image]

The GPS location and altitude is reported, though. So some communication takes place. In the GXAirCom GUI the data is all there:

[image]

So it appears the hardware is okay and GXAirCom can read out the values just fine, but doesn't transmit them to stratux in the correct way.

If you need further information or logs, please let me know.

Hardware:
RPi 4B
PMU: AXP2101
MCU: ESP32
GNSS: u-blox NEO-M8N
LoRa: SX1261

Software:
Stratux v1.6r1-eu032 (ff1f01dc41)
GXAirCom-v5.3.5-stratux-eu029
build-date: Jul 10 2023 05:46:54

VirusPilot, 2024-10-25:
No worries, your T-Beam is correctly providing GPS data to Stratux as you can see with "3D GPS" in your second picture. GXAirCom does not provide detailed satellite info ($GNGSV), therefore there is nothing to display in Stratux.

mpfusion, 2024-10-25:
Thanks for the blazing fast reply.

But this begs the question “why doesn't it do that”? Since the GPS button it not lit, stratux seems to expect that info (GPS based AHRS is unavilable) and the hardware provides it (as can be seen in the GXAirCom GUI) and SoftRF and ogntracker also send that info.

VirusPilot, 2024-10-25:
Well, it is up to the maintainer of this repo which GPS sentences are provided. I might consider creating a Stratux-compatible fork which would resolve this issue.

rvt, 2024-10-25:
You properly need this one : https://github.com/rvt/GXAirCom <https://github.com/rvt/GXAirCom>

The build scripts in stratux are configured here : https://github.com/b3nn0/stratux/blob/master/Makefile#L90 <https://github.com/b3nn0/stratux/blob/master/Makefile#L90>
It’s simple to setup the github actions and create a release, see : https://github.com/rvt/GXAirCom/actions <https://github.com/rvt/GXAirCom/actions>
This way Adrian can generate stratux versions together with your version of GxAirCOm.

The guy behind GxAircom is super friendly and helpfull if you need something special or have some ideas.

I don’t have time to maintain this combination any longer

Cheers,
Ries

mpfusion, 2024-10-25:
> You properly need this one : https://github.com/rvt/GXAirCom <https://github.com/rvt/GXAirCom>

I used the version supplied by stratux, which seems to be that very repository, if I understand it correctly. There's no mention of the differences between `https://github.com/rvt/GXAirCom` and `https://github.com/gereic/GXAirCom` in the README. If that's the version supplied by stratux (correct me if I'm wrong), I should already have that version, no? Else, what's the deal with that version? How does it differ? Does it fix the GPS issue?

> The build scripts in stratux are configured here : https://github.com/b3nn0/stratux/blob/master/Makefile#L90 <https://github.com/b3nn0/stratux/blob/master/Makefile#L90> It’s simple to setup the github actions and create a release, see : https://github.com/rvt/GXAirCom/actions

Before building a custom version, I think I'd first raise an issue on the stratux issue tracker to report that their supplied version is misconfigured.

>  I don’t have time to maintain this combination any longer 

Sorry to be ignorant (I'm a new user, bought the device two days ago), but are you suggesting that GXAirCom (which is listed as “experimental” on the stratux site) is deprecated now? The `gereic/GXAirCom` repo seems to be active as far as I can see.

To get back to the issue: @VirusPilot  seems to indicate that GXAirCom could emit the required messages. Instead of creating a fork what would be the downside of enabling those messages by default? Or if that's not desirable, adding a configuration option to enable them on request?

VirusPilot, 2024-10-27:
> To get back to the issue: @VirusPilot seems to indicate that GXAirCom could emit the required messages. Instead of creating a fork what would be the downside of enabling those messages by default? Or if that's not desirable, adding a configuration option to enable them on request?

If I find time, I can make an upstream PR to add a dedicated platformIO target for Stratux purposes.

VirusPilot, 2024-11-02:
As promised, here is my PR: https://github.com/gereic/GXAirCom/pull/173

mpfusion, 2024-11-02:
Thanks a lot for the quick implementation. I'll test it as soon as it's merged.

VirusPilot, 2024-11-03:
Made some further cosmetic changes and created the respective Github Actions commands so that now T-Beam binaries for Stratux use can be found under https://github.com/VirusPilot/GXAirCom/actions. Just click on the latest workflow result und download the GXAirCom zip file. You will find all necessary binaries inside, together with a partition mapping file.

mpfusion, 2024-11-04:
I tested the binary and it works like a charm. Position and satellites are reported and GPS button is lit. All seems to work. 

One minor thing: The GPS update frequency is 1Hz. According to the [Stratux web site - T-Beam-Firmware](https://github.com/b3nn0/stratux/wiki/TTGO-T-Beam-Firmware-for-Stratux) GXAirCom should be capable of sending 5Hz updates for better backup AHRS performance. It also says “free configuration”, but I fail to find a menu item to select the frequency.

Does it make sense to rise the GPS update frequency to 5Hz for the Stratux build?

Thanks a lot for the work. After this is merged in GXAirCom I'll create an issue on the Strtux tracker to pull in the new version, so the binary ends up in Stratux.

VirusPilot, 2024-11-04:
Good point, 5Hz should work easily. Let me look into it ...

mpfusion, 2024-11-04:
I just checked again and it does now indeed report 5Hz, not 1Hz. I'm not sure if I got confused with the different firmwares (I did re-flash and tried with different bins), or it started out at 1Hz and then adapted to 5Hz. But now after running it for a while it shows 5Hz as it's supposed to. So I believe you can disregard that and there's nothing else to do. Sorry for the noise.

VirusPilot, 2024-11-04:
No worries, in fact I spotted the section in the code where 5Hz is actually set. One more thing: please note that my binaries are only based on minor changes to the upstream code while @rvt 's Stratux version had many more options, e.g. selecting the aircraft ID through the Stratux web interface. So my binaries can't easily be used as a drop in replacement for the Stratux repository. Maybe I can implement some of these features soon but for now you have to flash and configure the T-Beam manually as you already did ...

rvt, 2024-11-04:
@VirusPilot Feel free to take my changes.. They we there just to satisfy Stratux and give a similar feel of the ogn-tracker.

VirusPilot, 2024-11-04:
@rvt which one of your forks was the most recent?

rvt, 2024-11-04:
Based on this one : https://github.com/rvt/GXAirCom/tree/stratux-eu029

mpfusion, 2024-11-04:
Ok, then I understand what the difference between rvt's version and gereic's version is. And also what's the deal behind [“Hoever we use a slightly modified version of GxAirCom and Stratux comes with a release of that.](https://github.com/b3nn0/stratux/wiki/GxAirCom-Integration)[sic]”.

I personally don't need tight Stratux GUI integration. However, since the code has already been written, it would be a shame to remove it. I don't know either project well enough to comment if it's feasible to merge the Stratux GUI integration upstream. But from an outside perspective this would probably make sense.

mpfusion, 2024-12-11:
Can this issue be closed now? AFAICS pull request #173 has been merged, not?

gereic, 2024-12-11:
Bug is fixed and testet with stratux.

closed 2024-12-11 by gereic