gereic/GXAirCom #174: Lilygo T3 s3 epaper compatibility

Martenz, 2025-01-14:
I'm receiving this new product from lilygo 

LILYGO® T3S3 E-Paper ESP32-S3 
LoRa SX1280 SX1262 SX1276

[image]

Anyone that already tested it? 
Any hint on configuring the GPIOs?

If not, I'll try to make it work. I love e-paper displays for paragliding cockpits; they offer perfect readability once you protect them with a matte shield.

Martenz, 2025-01-23:
Update:

I managed to get it partially working using a dedicated platform environment, along with some code modifications. You can see the differences in this feature branch.

[https://github.com/Martenz/GXAirCom/tree/feature/t3s3epaper](https://github.com/Martenz/GXAirCom/tree/feature/t3s3epaper)

What is still missing/not working:

* epaper 2.13" it works with GxEPD2_213_B74 but it is not refreshing data it seems partial mode is not working properly. 
* Lora sx1262 almost ok getting this init error or after a while but only when connected continuously to wifi
`[LoRa.cpp:284] SPItransfer(): error writing state=10 cmd=0X80`
* external GPS antenna, using a beitian bn180 on GPIO tx43, rx44, PPS39 (QwiiC 4 pin) works meaning it sends data to BLE or serial output but no data is displayed in the "info.html" web page. 
* BLE services working with XCTrack Android, not recognized with SeeYou Navigator App IOs

Anyone that can help on fixing some of the above errors ? 

Thanks, Martino

Martenz, 2025-02-02:
Update:

Solved almost all previous errors. Now on my feature branch it works and I modified slightly the screen.cpp to display correctly data because the eink is a 122 by 250 instead of 128 by 290.

This esp32 s3 has only 4MB flash so logger won’t fit yet.

Main issue still there is that I use an external gps specifically a beitian bn180 which doesn’t has pps pin.
If I don’t define PinPps then eink is never updated with data, if I set the PinPps to any of available gpio then I have to touch or change periodically the high low value of the gpio in order to see data updated on eink.

Anyone that can help me understand what this pps should do or how to set a gps properly without pps pin config?

Thanks

Martenz, 2025-02-05:
Update:

I fixed the issue where apparently no GPS data was received when setting GPS pins only for rx and tx (PinPPS = -1).

T3S3 doesn't have a GPS onboard, I just attached a GPS module to the Qwiic port that does not support then PPS.

To fix the error I had to comment this line:

[nmea.clearNewMsgValid();](https://github.com/gereic/GXAirCom/blob/d137c26b8ed9abdb54d17c694a0d3cd5f82ef1f7/src/main.cpp#L4932)

But I don't understand exactly why this clear command is there. Any help on this?

Now I have almost everything working except for logging that won't fit in this 4Mb flash, I'll might try to use compression option with platform build flags and try to add some code exclusion where possible only for this board to minimize libraries and code used to fit the logger (as far as we have on this board an SD module).

feel free to check and comment what I come up with in  [my fork / feature branch](https://github.com/Martenz/GXAirCom/tree/feature/t3s3epaper)

Thanks,

Martino.

slash-bit, 2025-02-06:
Good progress  👍 .
I would have though PPS is not that important for GPS operation. I have external  GPS without PPS, but not on GXAirComm,  on SoftRF. 
GXAirComm also works without GPS attached, when using app XCGuide (Android) connected via BLE. If you enabled BLE in the app it should autoconnect.  The app would send GPS nmea to GXAirComm if you select option "Send GPS to Fanet" in the app.
The app also configures some other parameters on GXAirComm and ints integrated nicely. 
If you kept the original ble code, then it should connect.
That could be your work around to understand what is causing display not to update. 
I don't have T3S3 epaper to test. But it's an interesting project. 

[image]
[image]

Martenz, 2025-02-06:
> Good progress  👍 .
> I would have though PPS is not that important for GPS operation. I have external  GPS without PPS, but not on GXAirComm,  on SoftRF.
> GXAirComm also works without GPS attached, [...]

Well with last modification it works just trying to understand what that clear NMEA sentences was there for.

I need to use it in AIRMODULE and send BLE data to XCTRACK or SEEYOU app because I want to use it as my main gps vario and fanet/flarm device for paragliding. 
T-Beam plus 2.9 eink would be an alternative but there is no sd and the board it's quite big. With this board I can fit everything in a smaller device that suits better for a paragliding cockpit.

I'm trying to keep the code original as much as possible plus keep all other boards compatibility, I added just a separate platformio env dedicated and few #ifdef to manage modified code for this specific board.

Martenz, 2025-02-22:
Got it work smoothly now also with logger igc files. 

I mostly edited the eink data displayed adding SD icon and recording status plus fanet and legacy RX TX status.
I made my own wifi / webhelper and used customized web pages, all  back to spiff data folder to reduce flash size and fit logger scripts finally in this 4Mb flash size. 

You can have a look in my fork and feature branch [here](https://github.com/Martenz/GXAirCom/tree/feature/tzi) 
I will try to keep the feature branch updated with latest core GxAircom code release. 

This is the end result:

[image]

3d printed TPU case which fit:

* a beitian bn-180 gps module
* a pam8302 amplifier
*  a 8ohm 1W speaker 
* touch sensor (button for volume)
* 3000mA battery