gereic/GXAirCom #176: Weatherdata over Heltec Lora V3

Moltobene3, 2025-01-25:
Hi,
is it possible to make a Weatherstation with windspeed and winddirection with teh Heltec Lora V3 over Fanet?
I still try, but i realy dont know the right Pin. 

[image]
[image]

nepat, 2025-01-25:
I have not seen the wind sensor on your picture being supported. But it is definitely possible with the Heltec Wireless Stick Lite V3 and wind sensors from Davis. I have built many of these stations. PCB file and parts list are included below.

[GXAirCom218650.zip](https://github.com/user-attachments/files/18548005/GXAirCom218650.zip)

[Parts List two 18650.ods](https://github.com/user-attachments/files/18548007/Parts.List.two.18650.ods)

nepat, 2025-01-25:
The end result will look like this.

[image]

[image]

[image]

[image]

[image]

gereic, 2025-01-27:
At the moment, weather-station for that board is not supported.
I can do this easy, cause I only have to define the pins for the anemometer.
If you like to do so, please answer and I will prepare it.

slash-bit, 2025-01-27:
This is a good idea, support for simple analogue anemometer and analogue direction  vine would be good.  I have exactly the same device  from weather station from Maplin. And was thinking coding it into GXAirCom. But better if its supported natively. It would a good portable weather station on a takeoff, "electronic windsock" to say.

Moltobene3, 2025-01-27:
Hallo und guten morgen. Ja das wäre super da ich 3 Wetterstationen von Skytraxx habe und die gerne auf den Heltec Lora V3 (die habe ich schon) umbauen würde. Der verwendete Windmesser ist von Misol WH-SP-WD und misol SP-WS

gereic, 2025-02-01:
Ich hab die Pins für das Anemometer so vergeben:

Bitte die aktuelle Version aus dem Git herunter laden.

******** BARO *********
SDA 2
SCL 3 

******** Anemometer (Davis 6410) *********
WINDDIR_PIN 33
WINDSPEED_PIN 34  (use external Pull 4,7k)

slash-bit, 2025-02-01:
Great, thanks. I will compile the latest version from Github and test.

Moltobene3, 2025-02-04:
Windspeed habe ich soweit hinbekommen das vernünftige daten gesendet werden, aber Windrichtung tut sich leider nichts

[Weather_Sensor_Assembly_Updated.pdf](https://github.com/user-attachments/files/18653215/Weather_Sensor_Assembly_Updated.pdf)

gereic, 2025-02-15:
Ich müsste für dieses Anemometer etwas eignes machen, da die Windrichtung nicht analog wie beim Davis ist.
Die Frage ist, wer aller so ein Anemometer nimmt.
Am einfachsten wäre es einfach ein DAVIS anzuschließen.

nepat, 2025-02-15:
In der Schweiz könnte ich Grundplatinen für Heltec Wireless Stick Lite V3 und Davis fertig @Materialkosten bereitstellen. Müsstest nur noch die Komponenten auf die Platine löten. Wichtig ist dabei das ganze Umfeld (Blitzschutz, automatischer Reset über RTC) nicht zu vernachlässigen. Bloss mit Heltec Platine und ein paar Widerständen hat man zwar rasch etwas zusammengesteckt, das funktioniert in der rauen Aussenwelt ohne Zusatzmassnahmen jedoch nicht dauerhaft zuverlässig.

VirusPilot, 2025-02-15:
> Ich müsste für dieses Anemometer etwas eignes machen, da die Windrichtung nicht analog wie beim Davis ist. Die Frage ist, wer aller so ein Anemometer nimmt. Am einfachsten wäre es einfach ein DAVIS anzuschließen.

DAVIS sendet wohl auch über 868MHz und es gibt schon eine Betaversion der OGN Glidernet Lib die DAVIS Wetterdaten übernimmt und an das OGN APRS weiterleitet.

Moltobene3, 2025-02-15:
oke, schade. Wie schon oben geschrieben, habe ich 3 Wetterstationen von Skytraxx die aber nicht mehr richtig funktionieren, und die haben eben dies Sensoren verbaut. zum wegschmeißen wäre es mir zu schade, da ja sonst alles funktioniert bis auf das Fanetmodul. preislich sind die auch in der Anschaffung bei ca 90€ und der Davis wäre das 3  bis 4 fache.

gereic, 2025-02-15:
ok, verstehe.
Dann müssen wir es so machen, dass wir das gemeinsam lösen.
Da ich keine solche Wetterstation habe, brauch ich deine Hilfe.
Ich werde eine Version vorbereiten und du musst mir dann die Werte von der Ausgabe senden.
Dann wird das funktionieren.

nepat, 2025-02-15:
Wichtig ist eine Echtzeituhr zu verbauen welche über den Reset Ausgang von aussen regelmässig einen Hard-Reset der Heltec Platine auslöst. Sonst kann es (besonders bei starken Gewittern) vorkommen dass die Signaleingänge blockieren und man dann an der Station manuell einen Reset auslösen muss. Absicherung aller Signaleingänge über TVS Dioden ist ebenfalls eine essentielle Schutzmassnahme. Heltec Lora V3 haben keine Abschirmung über den Komponenten, ein überflüssiges Display sowie keinen Anschluss für eine externe WiFi Antenne. Lora V3 deshalb nicht allererste Wahl für ein Hardwaredesign.

Moltobene3, 2025-02-15:
das wäre super

gereic, 2025-02-16:
Ich habe eine neue Version gebaut. V8.1.0
Hier kannst du im Webinterface bei dem Anemometer Winsol auswählen.
Bitte parallel zum Windrichtungssensor einen 10kOhm pull-up-widerstand einlöten.
In der seriellen Konsole, solltest du alle 2 sek. eine Ausgabe haben.
("[155331][I][Weather.cpp:250] checkAneometer(): analog-value of wind-vane:1023").
Hier bitte ganz langsam die Windrichtungsanzeige im Kreis drehen.
Dann sendest du mir die Log-Ausgabe.

P.S.: Die windgeschwindigkeit sollte schon passen. Bitte kontrollieren.

Moltobene3, 2025-02-18:
ok danke,

habe ich installiert, muss aber gleich vorweg sagen das ich mit Programieren und dergleichen überhaupt keine Ahnung habe.

schau mal bitte ob du mit dem was anfangen kannst, sonst würde ich sagen ich würde dir den Sensor zuschicken zum einstellen

gereic, 2025-02-19:
Ich denke es wäre am besten, du sendest mir den Sensor (Windmesser + Windrichtung).
Schick mir ein E-Mail an geichler78@gmail.com, dann sende ich dir meine Adressdaten.

gereic, 2025-02-21:
> [image]
> 
> ok danke,
> 
> habe ich installiert, muss aber gleich vorweg sagen das ich mit Programieren und dergleichen überhaupt keine Ahnung habe.
> 
> schau mal bitte ob du mit dem was anfangen kannst, sonst würde ich sagen ich würde dir den Sensor zuschicken zum einstellen

Danke für das Paket ;-)

Ich habe den Misol-Anemometer getestet und funktioniert soweit.
Der SHT20 hinten am Gehäuse kann auch angeschlossen werden.

[image]
SDA --> 2
SCL --> 3
Bei dem Lora V3 musst du noch 2.2kOhm pull-up-widerstände hinzufügen

Anemometer:
winddir --> 6
windspeed --> 7
Hier werden 10kOhm Pullup und 100nF Kondensatoren benötigt.

Ich denke aber auch, dass es am besten ist, du nutzt ein Board mit Heltec Wireless-Stick v3.
Ich könnte dir hier eines zukommen lassen.
Hast du noch ein Anemometer zum testen ?
Wäre super, wenn du es auch testen könntest und mir Bescheid gibts.

Moltobene3, 2025-02-23:
Ich habe leider momentan nix da zum Testen, da die anderen Stationen noch aufn Berg sind. kann da erst wieder hin wenn der Schnee weg ist.
Wenn du ein Heltc Wireless-Stick übrig hast dann bitte gern.

gereic, 2025-03-01:
Habe ein paar Stück bestellt.
Wennst keinen Stress hast, lege ich dir einfach einen bei, wenn Sie da sind und sende dir die Wetterstation wieder zurück.
Oder wir machen es so, dass du zuerst mit einer Station vom Berg testest, dann kann ich, sollte es Probleme geben, bei mir auch noch testen.

nepat, 2025-03-01:
In sehr naher Zukunft wird es für den Heltec Wireless Stick Lite V3 eine Trägerplatine in SMD Technik geben welche man günstig und komplett bestückt fremdfertigen lassen kann. Die Platine wird Solarlader, Echtzeituhr und Blitzschutz gleich auf der Platine integriert haben. Format etwa 6.2 * 8.5 cm

Moltobene3, 2025-03-01:
ja ich hol die Station vom Berg und werd testen.

Moltobene3, 2025-03-13:
die Sensoren funktionieren. Winddir und winspeed passen. Was ich aber festgestellt hab, das wenn die station an die Rasperry OGN station sendet, die Station auf https://www.gliderradar.com/ und https://breezedude.de/?lat=46.8600&lon=13.3947&z=11.9&id=26080 zu sehen ist. aber wenn sie auf die GXaircom Bodenstation sendet die daten auf https://breezedude.de/?lat=46.8600&lon=13.3947&z=11.9&id=26080 nicht mehr empfangen werden.

@nepat 
meintest du eventuell das system? 
https://de.elv.com/elvwissen/elvwiki/das-elv-wetterboard-wetterstation-massgeschneidert/

nepat, 2025-03-13:
Es wird eine fixfertige Trägerplatine für Heltec Stick Lite V3 werden welche alle Anschlüsse und Komponenten im Umfeld bereits enthält. D.h. Solarlader, Echtzeituhr, Blitzschutz. Man wird ausser der Stiftleiste am Heltec nichts löten müssen. Komfortable Konfiguration für alle unterstützten Windmesser über DIP Schalter. In etwa zwei Wochen sollte ich den ersten Prototypen testen können.

[image]

Moltobene3, 2025-04-28:
@nepat  wieweit sind deine Tests für die Platine?

gereic, 2025-05-04:
@Moltobene3 
Hab heute die neue Platine mit deinen Komponenten getestet und es funktioniert soweit alles.
[image]
Modem für Internet ist keines verbaut.
die Platine würde sogar ins originale Gehäuse passen, einzig die Batterie ist zu dick.

Moltobene3, 2025-05-04:
wow das sieht gut aus. Modem brauchts ja keines wenn die daten über fanet gesendet werden? und die Solarzelle reicht da aus?

gereic, 2025-05-04:
Die Solarzelle würde ich gegen ein anderes Panel mit mehr Leistung tauschen.
Auf der Platine kann man die Spannung des Panels über DIP-Switch einstellen.
Ich musste nur noch 2 Widerstände dazu löten, da der SHT20 einen Pull-up-Widerstand von ~2kOhm braucht.
Wenn du selbst eine Platine herstellen möchtest kann ich dir noch ein paar testplatinen senden.

Moltobene3, 2025-05-05:
ja bitte das wäre echt super

gereic, 2025-05-11:
@Moltobene3 
Magst du mir noch einmal deine Adressdaten senden ?
Dann kann ich das Paket morgen aufgeben.