gereic/GXAirCom #179: GXAircom transmitting on wrong frequency

sirapm, 2025-02-18:
Been trying to get gxaircom to work on Heltec Lora V3 and lilygo tbeam s3core with little success in New Zealand.
I finally fired up a spectrum analyzer and found that both Flarm and Fanet is transmitting on 921Mhz ish.
New Zealand is using 869.25Mhz for Flarm (and I guess Fanet).
Can this be fixed please? Maybe add manual / override setting for frequency?
Cheers Martin

sirapm, 2025-02-18:
And just noticed it is shifting frequence ever down (a mhz or so ever know and then).

sirapm, 2025-02-18:
Another useful thing would be to add "current frequency" to the info page off the web interface so one can easily see what frequency that is in use.

sirapm, 2025-02-18:
[GXAirCom](https://github.com/gereic/GXAirCom/tree/master)/[lib](https://github.com/gereic/GXAirCom/tree/master/lib)/[FLARM](https://github.com/gereic/GXAirCom/tree/master/lib/FLARM)
/v7.txt
This document is wrong. Zone 3, New Zealand is using 869.2 (and some other online resources are saying 869.25MHz)
See https://www.flarm.com/wp-content/uploads/2024/04/FTD-078-PowerFLARM-Fusion-User-and-Maintenance-Manual.pdf
section 1.5
The following frequencies are used within the specified areas.
Area Frequency
Africa 868.2 – 868.4 MHz
Australia 917.0 – 926.6 MHz
Europe 868.2 – 868.4 MHz
Israel 916.2 MHz
New Zealand 869.2 MHz
North America 902.2 – 927.8 MHz
South America 917.0 – 926.2 MHz
Can this easily be updated and fixed?
Cheers Martin

VirusPilot, 2025-02-18:
Here is how SoftRF is defining the areas:

	SOFTRF_BAND_EU   = 1, /* 868.4 MHz band */
	SOFTRF_BAND_US   = 2, /* 915 MHz band */
	SOFTRF_BAND_AU   = 3, /* 921 MHz band */
	SOFTRF_BAND_NZ   = 4, /* 869.250 MHz band */
	SOFTRF_BAND_RU   = 5, /* 868.8 MHz band */
	SOFTRF_BAND_CN   = 6, /* 470 MHz band */
	SOFTRF_BAND_UK   = 7, /* 869.52 MHz band. Deprecated - treated as EU */
	SOFTRF_BAND_IN   = 8, /* 866.0 MHz band */
	SOFTRF_BAND_IL   = 9, /* 916.2 MHz band */
	SOFTRF_BAND_KR   = 10 /* 920.9 MHz band */

and here the OGN definitions:

          if(Plan==2) { BaseFreq=902200000; ChanSepar=400000; Channels=65; } // USA
     else if(Plan==3) { BaseFreq=917000000; ChanSepar=400000; Channels=24; } // Australia and South America
     else if(Plan==4) { BaseFreq=869250000; ChanSepar=200000; Channels= 1; } // New Zeeland
     else             { BaseFreq=868200000; ChanSepar=200000; Channels= 2; } // Europe

sirapm, 2025-02-18:
Thanks. 
How do we get this in to GXAircom? 
Cheers
Martin

sirapm, 2025-02-18:
If the lib/flarm/FlarmRadio.cpp is in use, I guess that's the issue. 
Just need a new a few new rows of code on line 183,
  }else if (zone == 3){
    //Zone 3: f0=869.25, nch=1
    *frequency = 869.25;
    *channels = 1;

mikechristiansen, 2025-02-18:
@sirapm, where are you located in NZ? I have a few GxAirCom nodes around the wairarapa. Mostly im not too concerned about FLARM because we have receivers covering this area already, but I have played with this with limited sucess. Of note, FLARM packets are not forwarded from remote stations to internet connected stations (no forwarding of FLARM on ground stations). 
Would be good to hear from you. Cheers!  zero two seven 4164649

sirapm, 2025-02-18:
@mikechristiansen I'm in Masterton... I'm in WWGC gliding club. 
I have two objectives. Replace the "not so good OGN" Flarm receiver with this and feed ogn.
Second objective is two fold. In my glider, I'm keen to have a Fanet RX/TX (think that is working). I do from time to time come across para/hang gliders. Hopefully, some of them have Fanet (never seen any of them with Flarm). Even though one is keeping a good lookout, never hurt to also electronically broadcast my presence... Want to give them as much space as possible. 
And the second part would be to see if I can get Fanet RX -> Flarm Tx broadcast to help other gliders see Fanet devices. 
Basically, I'm quite sure Fanet is working, but not Flarm as it is working on the wrong frequency in New Zealand.

mikechristiansen, 2025-02-19:
@sirapm I am also in Masterton, I would be one of the Paragliders you would have come accross! I have GxAirCom stations running on Jury hill, Kourarau and one on a dairy farm in Carterton acting as weather stations. I also have 2 internet connected stations at my place in masterton and one in Carterton feeding FANET weahter and tracking to OGN, WeatherUnderground, Windy and PureTrack. These are setup only to do FANET at this stage, and effectively covers the entire valley, ive flown down on the south coast and tracking works just above ridge height. 

My home node is setup for FLARM RX, but even with the changes mensioned above (ive modified the code) it wasnt working. 

Interested in your ideas about FANET-RX to FLARM too, surely that would help with safety!

sirapm, 2025-02-19:
Cool. 
Do you have a pointer to docs about how to recompile gxaircom? Not sure if/when the base code will be updated to use the correct frequency. Keen to have a play with recompiling but not my normal domain...
Our current flarm and ADS-B also cover most of the flat area in the Wairarapa. Would love to be able to get some receivers with solar, battery and cheap 4g devices a bit further up north.

VirusPilot, 2025-02-19:
@sirapm I have compiled a patched version in my fork, you can download it from my Github actions page: https://github.com/VirusPilot/GXAirCom/actions/runs/13407268234.

gereic, 2025-02-19:
Has anyone double-checked the frequency with a real flarm-device ?
I can adjust it easy, but want to be sure, that it is the right frequency.

VirusPilot, 2025-02-19:
@gereic just one more question to the prior mentioned code in `lib/flarm/FlarmRadio.cpp`: EU seems to allow two channels but what does `*channels = 0;` for zone 1 (EU) actually mean? Shouldn't it actually be `*channels = 2;`?

PIDIENNE, 2025-02-19:
> Has anyone double-checked the frequency with a real flarm-device ? I can adjust it easy, but want to be sure, that it is the right frequency.

I did a quick test with a Flytec 6030 FLARM (set on EU frequency I guess). It detected the GXAircom FLARM with FW 7.0, but apparently not with 8.0

gereic, 2025-02-19:
> > Has anyone double-checked the frequency with a real flarm-device ? I can adjust it easy, but want to be sure, that it is the right frequency.
> 
> I did a quick test with a Flytec 6030 FLARM (set on EU frequency I guess). It detected the GXAircom FLARM with FW 7.0, but apparently not with 8.0

I mean in Newzealand.

slash-bit, 2025-02-19:
Hey. See if that helps. I recorded those a while ago using XC Tracer Maxx 2 EU region ( I guess that is official enough device)

https://github.com/user-attachments/assets/7323e523-aa36-497b-abbb-c0f32cd3b4dc

https://github.com/user-attachments/assets/3fa04187-b382-47a3-a175-105d81ba96c0

https://github.com/user-attachments/assets/eb25faf8-d60d-49eb-9f27-8968cda41611

gereic, 2025-02-19:
That EU and not NZ.
So it doesn't help for the problem in NZ

sirapm, 2025-02-20:
NZ is fir sure on 869.25. My flarm mouse is sending on 869.25, a long with all other flarm devices in the club. After modifying the flarm code (as shown above) to use 869.25 other flarm receivers (OGN receivers) are picking up my t-beam fine. However, not my heltec receiver. Even when hardcoding 869.25 regardless of zone, the heltec V3 gxaircom doesn't pick up the gxaircom transmitter. Not sure what to try next.

gereic, 2025-02-20:
> NZ is fir sure on 869.25. My flarm mouse is sending on 869.25, a long with all other flarm devices in the club. After modifying the flarm code (as shown above) to use 869.25 other flarm receivers [...]

I created a new Version with the modifications for NZ. Can you try it and give me some feedback please ?

mikechristiansen, 2025-02-22:
Good flying day currently. 
Ive jsut tried to run a test to RX FLARM from a glider close to my location.
Using a known good YAGI pointed at it 10km out no good. 
Im getting the correct FLARM zone on boot logs, but zero RX FLARM or aircraft showing. 

I assume the targeted glider was FLARM equipped? @sirapm you might be able to comment here? 

Hardware is a Tbeam, ive set it up to FLARM RX only. 

[image]

--- Terminal on /dev/cu.usbserial-01ED9EAB | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[I][main.cpp:4148] checkGPSBaud(): gps: baudrate with 9600 ok
[I][main.cpp:4125] checkGPSBaudrates(): gps connection ok with baud 9600
[I][main.cpp:4481] taskStandard(): setup PPS-Pin for GPS
[I][fmac.cpp:528] begin(): sck=5,miso=19,mosi=27,ss=18,reset=23,dio0=26,gpio=-1,chip=2
[I][LoRa.cpp:478] begin(): sx1276-Version 18
[I][fmac.cpp:553] begin(): LoRa Initialization OK!
[I][FanetLora.cpp:77] begin(): myDevId:084C4C
[I][FanetLora.cpp:90] setGPS(): disable GPS PPS
[I][fmac.cpp:1008] setRegion(): set Lora-Region to AU920
[I][FlarmRadio.cpp:159] flarm_get_zone(): Zone 3: New Zealand (east of 160E)
[I][fmac.cpp:1040] setRegion(): zone=3,freq=869.2,channels=1
[I][LoRa.cpp:1215] startReceive(): start calib image
[I][main.cpp:1376] setupWifi(): setup Wifi
[I][main.cpp:1386] setupWifi(): setup Wifi ready !
[I][main.cpp:3523] setWifi(): switch WIFI ON
[I][main.cpp:3529] setWifi(): set CPU-Speed to 240MHz
I NimBLEDevice: BLE Host Task Started
I NimBLEDevice: NimBle host synced.
[I][ble.h:178] start_ble(): Starting BLE
[I][ble.h:187] start_ble(): Waiting a client connection to notify...
[I][main.cpp:1311] WiFiEvent(): wifi ready
[I][main.cpp:1314] WiFiEvent(): station start
[I][main.cpp:3541] setWifi(): Try to connect to WiFi 
[I][main.cpp:1366] setupSoftAp(): Set soft-AP settings
[I][main.cpp:1311] WiFiEvent(): wifi ready
[I][main.cpp:1343] WiFiEvent(): soft-AP start
[I][main.cpp:1348] WiFiEvent(): soft-AP stop
[I][main.cpp:1343] WiFiEvent(): soft-AP start
[I][main.cpp:1314] WiFiEvent(): station start
[I][main.cpp:1314] WiFiEvent(): station start
[I][main.cpp:1323] WiFiEvent(): station connected to AP

mikechristiansen, 2025-02-22:
Further to the above, I pulled out an SDR and could see that if I enabled FLARM-TX I could see packets on 869.25 but as soon as I turned that off, nothing. Ive been told by multiple people that FLARM runs on 869.25 but I cant seem to receive anything.

sirapm, 2025-02-23:
Loaded the 8.1.0 with NZ modifications. However, frequence is listed as wrong.
868.2
From my spectrom analyzer. GXAircom is the one on the left and my softrf is on the right.

[image]

gereic, 2025-02-23:
> Loaded the 8.1.0 with NZ modifications. However, frequence is listed as wrong. 868.2 From my spectrom analyzer. GXAircom is the one on the left and my softrf is on the right.
> 
> [image]

In the info-page you should see the Fanet/Flarm-Frequency.
What is schon there ?
Please send me your location (lat,lon).

gereic, 2025-03-01:
Was anyone able to test the new version ??
Feedback would be welcome.

mikechristiansen, 2025-03-01:
Tested FLARM TX today. Seems to be working well in NZ. Im unable to test RX however, as I have no near by aircraft to receive. 

[image]

[image]

gereic, 2025-03-02:
perfect, so I will close this issue.

sirapm, 2025-03-03:
Not sure why closed. Tx works, but not receive, RX...
Cheers 
Martin

gereic, 2025-03-04:
Sorry, didn't know, that RX not working.
Do you have a Flarm-Device, which can TX (like skytraxx or xctracer) ?

sirapm, 2025-03-06:
I have tested with gxaircom, flarm mouse, and rfsoft. Nom are picked up by gxaircom. Even when I hardcoded the frequency to 869.25. So not sure where the problem is. Unless RX and tx is managed by different source code.

sirapm, 2025-03-06:
By the way, thanks for the enormous amount of work. Truly inspiring!

gereic, 2025-03-06:
Only to be sure, you have testet it with softrf on 869.25 and GxAircom doesn't receive SoftRf ?
I have also a SoftRf-Device, so I can test that case.

sirapm, 2025-03-07:
I've tested with transmitting flarm with a native flarm mouse, softrf and gxaircom. All on 869.25 and none of them were picked up by gxaircom receiver. I've tested on both a heltec V3 and a lilygo supreme. Neither picked up any of the flarm messages.
I even hardcoded the frequency in the software to 869.25, regardless of zone. Still no received messages.

gereic, 2025-03-08:
> I've tested with transmitting flarm with a native flarm mouse, softrf and gxaircom. All on 869.25 and none of them were picked up by gxaircom receiver. I've tested on both a heltec V3 and a lilygo [...]

I can reproduce the error.
Is the Softrf-device receiving the flarm mouse ?

sirapm, 2025-03-08:
I can load Softrf on my Lilygo again and test next time I'm at the air
field. May not be for another week or two. :(
Cheers
Martin

gereic, 2025-07-16:
Can someone try the latest version v8.2.1
I modified the code and it is receiving and transmitting.
Tested with softrf-device.
Frequency is 869.2Mhz

closed 2026-01-03 by sirapm