gereic/GXAirCom #182: No gps fix with GPS module without PPS signal

hermannbach, 2025-02-27:
Identification of gps fix only works with PPS signal, may be with $PPS message. When having a gps module with none of them the gps fix is not identified correct.

Problem seems to be a confusion of the two variables tOldPPS  and gtPPS.

In main.cpp the triggertime of 1000 ms doesn't work. Here the corresponding code snippet.
` 
if ((tAct - tOldPPS) >= 1000){
          gtPPS = millis();
          ppsTriggered = true;
        }
      }
`
changed tOldPPS to gtPPS and it works for me.

A similar problem: when pps signal is connected and defined, after the timeout of 10 seconds that part is executed for 10 more seconds in every turn.
`if((tAct - tLastPPS) >= 10000){
//no pps for more than 10sec. --> clear GPS-state
....
`
A line 
`tLastPPS = tAct`
at the end of that part fixes the problem for me.
Because at the moment I didn't really understand that part in detail, I don't know if my fixes have any side effects!?

gereic, 2025-03-01:
So you are using a configuration without pps.
Without pps you can't use Flarm.

I will check that and let you know.

hermannbach, 2025-03-01:
Yes I have an older bluefly vario, connected to a Heltec V2.1 connected to a KOBO Glo. The bluefly Vario has no PPS. If FLARM isn't working, it isn't a big problem in this case.
Highly appreciate your awsome work on GXAirCom.
Last season most time I used a modified Version 5 GXAirCom on a Heltec V2.1 with pressure sensor MS5611, MPU6050 and luckily found good piezzo speaker connected to a KOBO mini as flight instrument. Worked like a charm.

muzzl3, 2025-04-26:
@hermannbach, I just wired up an old TTL bluefly vario onto one of the Heltec boards since I had one here and I've gotten used to the beeps the bluefly produces. 
I thought it would be worth mentioning for others that might find this, that once you know which version vario you have and find the schematic specific to that board/gps chip you can just take the PPS signal and attach it to whichever pin you have selected on your ESP32 board.

gereic, 2025-05-14:
With v8.2.1 it should work, please give me some feedback.