gereic/GXAirCom #189: Frequency Correction Option (max. +/- 30kHz)

VirusPilot, 2025-03-21:
This is just a feature request. Some T-Beam devices are quite off (-20.8kHz in case of my current T-Beam). This correction can only be verified with a reasonably well calibrated OGN station but I believe it is worth doing it.

gereic, 2025-03-23:
So the frequency-correction has to be applied generally for all protocols ?

VirusPilot, 2025-03-23:
I guess so, whenever a SX1262 or SX1276 is used.

gereic, 2025-03-23:
I will upload a new version, you have to set "settings view" to expert and you will find the setting "Frequency Correction (max. +/- 30kHz) [kHz]".
I have no calibrated SDR-Dongle here, so can you try it ?
New version is v8.2.0

VirusPilot, 2025-03-23:
Works like a charm, here in my case with a 20kHz correction, captured with my calibrated OGN station:

[image]

gereic, 2025-03-23:
Do you know, how I can build a calibrated OGN-Station ?
Want to check my stations.

VirusPilot, 2025-03-23:
I described it here in my script:

https://github.com/VirusPilot/ogn-pi34?tab=readme-ov-file#sdr-ppm-correction-mandatory-for-non-tcxo-sdrs

You can even further activate automatic fine calibration:

https://github.com/VirusPilot/ogn-pi34?tab=readme-ov-file#sdr-ppm-automatic-fine-calibration---100-ppm

VirusPilot, 2025-03-23:
Here is how it looks like on the OGN monitoring site:

RF.FreqCorr is the manual correction
Fine calib. FreqCorr is the automatic correction
RF.GSM.CenterFreq is the required reference signal

[image]

gereic, 2025-03-23:
> Works like a charm, here in my case with a 20kHz correction, captured with my calibrated OGN station:
> 
> <img alt="Image" width="661" src="https://private-user-images.githubusercontent.com/43483458/425820750-29fb329a-c3e9-4ad6-a3e4-bf97cdda570b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI3Mzg5NzAsIm5iZiI6MTc0MjczODY3MCwicGF0aCI6Ii80MzQ4MzQ1OC80MjU4MjA3NTAtMjlmYjMyOWEtYzNlOS00YWQ2LWEzZTQtYmY5N2NkZGE1NzBiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAzMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMzIzVDE0MDQzMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM5NWRjM2FhN2RhYmVmYTI5OWM3ODgwMzUzNzQ5NTI3OTgwZjBmNDZjMWI0ZDQ2MzYyOTE2YjYyNDJhY2NjZGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ggVwAuYVn2l6jBeBNkrvb_8pfAHIBlwZ63Mg4hWJoh4">

How do I get this output ?

VirusPilot, 2025-03-23:
Login into your OGN station and run the following command:

`nc localhost 50012 | grep FLR6`

If nc does not yet exist, then you need to install netcat:

`sudo apt install netcat` (Bullseye)
`sudo apt install netcat-traditional` (Bookworm)

gereic, 2025-03-23:
ok, thanks.
so far, can we close this issue ?

VirusPilot, 2025-03-23:
yes, thank you once again

closed 2025-03-23 by VirusPilot