gereic/GXAirCom #191: [Feature idea] repeater only

bsiege, 2025-03-28:
Could help in the Alps to give better coverage towards connected ground stations. Needs Lora Board only, no GPS needed. Small package with Solar Cell. Starts to repeat FANET packets up to maximal allowed duty cycle during daylight.  No more, no less.

nepat, 2025-03-28:
GXAircom software already does this and repeating is proving useful for example for mounting weather stations in locations that require a repeater hop to reach a base station. Here you have a prototype PCB with solar charger electronics and real time clock. The real time clock enables timed operation (for example sunrise to sunset) and from time to time performs a hard reset of the entire system. PCB will go on Github after completion of testing.

[image]

bsiege, 2025-03-28:
Ok, will test that soon. Have enough Hardware here. Wanted to be shure, that the Ground Station itself is absolute silent and transparent. No Name needed, no coordinates needed...