gereic/GXAirCom #194: Wakeup WiFi via gsm gateway

Oganisyan, 2025-03-30:
It would be good to have a way to activate WiFi via gsm(e.g., via sms). Otherwise, the module would have to be taken down from the mast for every reconfiguration.

rvt, 2025-03-30:
Alternative, you could send 'something' over the LORA frequencies. GxAirCom already does FANET so you could send a specially crafted message with some key.

nepat, 2025-03-30:
this function (wakeup with lora packet)is already implemented - is a great feature !

nepat, 2025-03-30:
Essentially you first set a Pin code on the receiver station. Then you can send command messages from any other GXaircom board by selecting the receiver ID, Pin and message type....and hit the send button. I like to carry a Heltec Lora32 in plastic case around with me for that purpose.

closed 2025-04-19 by Oganisyan