gereic/GXAirCom #195: V8.1 on T-Beam V0.7 Web Info says no GPS fix; but there is a fix and works in XCSoar

sbishop61, 2025-04-03:
V8.1 instaned on T-Beam V0.7.
It takes a lfetime to get a fix (no probolem with V5).
Then the PPS light falshes but the webinfo says no fix and hence no TX for Fanet or Flarm.
However when I look at the serial ouput there is a fix.
I connect to XCSoar via BLE and XCSoar reports there is a fix - and it is in the right place.

Is this a setting issue or a bug?

gx-nick, 2025-04-06:
Have you  clicked the "configure GPS" on the settings page of the web interface? This changes the enabled/disabled NMEA messages on the GPS module. I'm not sure about your device or if this applies broadly to the ublox modules but the off the shelf configuration on the chip is probably at fault. Try the configure GPS option while connected to the serial to see if this succeeds or otherwise if there's some issue writing the required settings to the GPS module.

sbishop61, 2025-04-06:
GX-Nick I did do that. It still didn't work.

I have now soldered a wore between the Neo-6M PPS pin and GPIO25 and made the required change to Main.cpp switch/case statement an built v8.2 - that works

Netxxl, 2025-05-13:
Could someone please add this fix to the official repo for the T-Beam V0.7 ?
Not everyone wants to compile their own version probably.
Soldering a wire to any GPIO Pin is not for everyone too, but doable.
Thanks!

gereic, 2025-05-14:
Bug should be fixed.
Please try v8.2.1
Please report if it is working or not.

sbishop61, 2025-05-14:
If I leave the soldered wire between the Neo-6M PPS pin and GPIO25; but use the stock noPSRam 8.2.1 firmware (no PPS pin defined), presumably that still tests the fix?

gereic, 2025-05-15:
> If I leave the soldered wire between the Neo-6M PPS pin and GPIO25; but use the stock noPSRam 8.2.1 firmware (no PPS pin defined), presumably that still tests the fix?

Yess.