gereic/GXAirCom #196: cannot update Firmware on T-Beam?

alexanderdross, 2025-04-15:
I tried to update the firmware of GXAirCom from "**GXAirCom-v5.3.5-stratux-eu029 build-date: Jul 10 2023 05:46:54**" to its latest version.
Since I am using the **T-Beam T22 V1.1 SX1262** - the following firmware was selected:
https://github.com/gereic/GXAirCom/tree/master/bin/T-Beam-S3Core/v8.2.0 

uploaded both, 1. firmware.bin file 2. spiffs.bin file. the progress bar changed to 100% and T-Beam rebooted. However, Loggin-in to the interface still shows the above build-date/ version.

What am I doing fron here?

sbishop61, 2025-04-15:
You are using S3 firmware when, I think, you should be using the psRam firmware.

You have tried to flash with firmware for an S3 chip, but the T-beam T22_V1.1does not have an S3 MCU.
Try with the psRam firmare under 8.2. Make sure that you have the locations correct per the image and do an erase first, then flash the firmware.

alexanderdross, 2025-04-15:
@sbishop61 awesome! that worked. thanks for your quick hint

closed 2025-04-15 by alexanderdross