gereic/GXAirCom #197: Support for WS85 anemometer.

mikechristiansen, 2025-04-20:
Hi,
Im planning on working on this issue and will work out how to do a pull request - will try my best! 

Looking to intergrate the WS85 serial message based anemometer that is supported by Meshtastic and Breezedude. 
https://github.com/thezenox/breezedude

The WS85 anemometer is an ultrasonic sensor, avalable very cheaply and requires a small modification to get serial output from it. It is self powered and seems very reliable. It provides wind direction, speed, tempreature and it looks like rain info as well! 
https://www.aliexpress.com/item/1005006903955376.html
https://www.youtube.com/watch?v=bpVVf8dRDc4

Im planning on using one of the wind pins currently in the code for a software serial port. Weather is sent by this anemometer at 2s intervals, with a baud rate of 115200. 

Example of the code used in meshtastic to praise the serial data below. 

 while (Serial2.available()) {
        // clear serialBytes buffer
        memset(serialBytes, '\0', sizeof(serialBytes));
        // memset(formattedString, '\0', sizeof(formattedString));
        serialPayloadSize = Serial2.readBytes(serialBytes, 512);
        // check for a strings we care about
        // example output of serial data fields from the WS85
        // WindDir      = 79
        // WindSpeed    = 0.5
        // WindGust     = 0.6
        // GXTS04Temp   = 24.4
        // Temperature = 23.4 // WS80
        // RainIntSum     = 0
        // Rain           = 0.0
        if (serialPayloadSize > 0) {
            // Define variables for line processing
            int lineStart = 0;
            int lineEnd = -1;

            // Process each byte in the received data
            for (size_t i = 0; i < serialPayloadSize; i++) {
                // go until we hit the end of line and then process the line
                if (serialBytes[i] == '\n') {
                    lineEnd = i;
                    // Extract the current line
                    char line[meshtastic_Constants_DATA_PAYLOAD_LEN];
                    memset(line, '\0', sizeof(line));
                    if (lineEnd - lineStart < sizeof(line) - 1) {
                        memcpy(line, &serialBytes[lineStart], lineEnd - lineStart);

                        ParsedLine parsed = parseLine(line);
                        if (parsed.name.length() > 0) {
                            if (parsed.name == "WindDir") {
                                strlcpy(windDir, parsed.value.c_str(), sizeof(windDir));
                                double radians = GeoCoord::toRadians(strtof(windDir, nullptr));
                                dir_sum_sin += sin(radians);
                                dir_sum_cos += cos(radians);
                                dirCount++;
                                gotwind = true;
                            } else if (parsed.name == "WindSpeed") {
                                strlcpy(windVel, parsed.value.c_str(), sizeof(windVel));
                                float newv = strtof(windVel, nullptr);
                                velSum += newv;
                                velCount++;
                                if (newv < lull || lull == -1) {
                                    lull = newv;
                                }
                                gotwind = true;
                            } else if (parsed.name == "WindGust") {
                                strlcpy(windGust, parsed.value.c_str(), sizeof(windGust));
                                float newg = strtof(windGust, nullptr);
                                if (newg > gust) {
                                    gust = newg;
                                }
                                gotwind = true;
                            } else if (parsed.name == "BatVoltage") {
                                strlcpy(batVoltage, parsed.value.c_str(), sizeof(batVoltage));
                                batVoltageF = strtof(batVoltage, nullptr);
                                break; // last possible data we want so break
                            } else if (parsed.name == "CapVoltage") {
                                strlcpy(capVoltage, parsed.value.c_str(), sizeof(capVoltage));
                                capVoltageF = strtof(capVoltage, nullptr);
                            } else if (parsed.name == "GXTS04Temp" || parsed.name == "Temperature") {
                                strlcpy(temperature, parsed.value.c_str(), sizeof(temperature));
                                temperatureF = strtof(temperature, nullptr);
                            } else if (parsed.name == "RainIntSum") {
                                strlcpy(rainStr, parsed.value.c_str(), sizeof(rainStr));
                                rainSum = int(strtof(rainStr, nullptr));
                            } else if (parsed.name == "Rain") {
                                strlcpy(rainStr, parsed.value.c_str(), sizeof(rainStr));
                                rain = strtof(rainStr, nullptr);
                            }
                        }

                        // Update lineStart for the next line
                        lineStart = lineEnd + 1;
                    }
                }
            }
            break;
            // clear the input buffer
            while (Serial2.available() > 0) {
                Serial2.read(); // Read and discard the bytes in the input buffer
            }
        }
    }

nepat, 2025-04-20:
There may be a better way than modifying hardware and running cables.

Fine Offset / Ecowitt make a large range of weather sensors that are sold under many sub-brands.

https://meshka.eu/Ecowitt/dokuwiki/doku.php?id=start

There is a library for the ESP32 that enables wireless connection to Fine Offset weather sensors

https://github.com/NorthernMan54/rtl_433_ESP

Fine Offset devices operate on two frequencies: 433MHZ and 864MHZ  This depends on the market (countries) into which the devices are sold. Western Europe uses 864MHZ on FSK modulation.

Support for the SX1262 appears to be not mainstream yet https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/80

Fine Offset sensors are very widespread and affordable. The newer sensors are all solar powered and wireless. It would be fantastic if the rtl_433_ESP library could be integrated into GXAirCom.

mhofner21, 2025-05-15:
Hey @mikechristiansen ,
I really like your idea – in general, support for UART-based sensors like the WS80, WS85, and WS90 would be a great addition. Have you made any progress with the implementation?

We’ve had great experiences with the WS80 at our airfield – an Ecowitt WS80 has been mounted there for over three years and is still running flawlessly. It hasn’t needed any maintenance, and it's still on the original batteries. I’ve had similar experiences at home, where I use RTL-SDR and rtl_433 to wirelessly receive the data and forward it via MQTT to Home Assistant. That works very well – but of course, I have no power limitations at home.

Florian from Breezedude confirmed to me on Facebook what I had already suspected: power consumption for continuous listening via radio is quite high. That’s why he chose to connect the WS80 via UART and a wired connection in his project.

@nepat , I really like your PCB design – I’d be very excited to test it with a UART-modified WS80.

Since we live in a flatland region, extreme weather conditions aren't really an issue for us.

Unfortunately, I lack the technical skills to add UART support to GXAirCom myself. I also find Matthias' project (connecting Bresser sensors to TTN) very interesting, but I believe it's better suited for ground stations where power efficiency isn’t critical.

That’s why I find both approaches promising:
– Extending GXAirCom with UART support for autonomous setups
– Using rtl_433 to integrate Ecowitt sensors in more powerful, stationary systems