gereic/GXAirCom #200: set custom HEX ID

alexanderdross, 2025-05-05:
As I use a transponder with ADS-B out, I want to use the same HEX ID for GxAirCom, such as for Flarm TX.
In the web interface at http://192.168.4.1/, I can view the HEX ID, but I cannot modify or update it.

Is there a plan to enable this feature in the future?

alxwolf, 2025-06-05:
@alexanderdross @gereic 

#186 killed this feature (I assume, accidentally).

Asked to get it back: https://github.com/gereic/GXAirCom/pull/186#issuecomment-2943643643

luar123, 2025-06-05:
That was not intended, sorry. I guess it could be reverted easily, then it wouldn't be standard compliant but shouldn't break anything.
Maybe a better solution would be to send a custom request from stratux? Just an idea....

Anyway, there is a more critical issue with ICAO hex ID: You can not use them for Fanet (you could but you shouldn't as it could lead to two devices having the same ID). So if you use this feature please don't enable Fanet TX.
There is an additional Fanet message type (not implemented by gxaircom) that can include the ICAO ID for exactly that purpose (not sure if this is used by any other device to match the IDs, though). So the ideal solution would be to implement this message type, add an additional field for an optional ICAO address and use ICAO or internal ID for Flarm TX, internal ID for Fanet Tx and send ICAO as additional message with Fanet.

alxwolf, 2025-06-05:
Yes that's "dirty" - but it worked :) 

Stratux [actually checks for 3 things](https://github.com/stratux/stratux/blob/df9f13de63641d7aa91f8450b60ea590e5e97993/main/tracker.go#L230): 

"GXAircom" in $PFLAV message, "$LK8EX1" message and the $PGXCF message. 

Sadly, I've spent a full afternoon and couldn't even find out why the firmware in `bin` directory  has UDP as output and whenever I compile locally I get BLE, and both variants don't have serial output and Beidou and Galileo satellite messages are disabled until "GPS config" is clicked... 

So, I'd rather leave this to someone who has a clue... would be great if we could [stratux' `tracker.go`](https://github.com/stratux/stratux/blob/master/main/tracker.go) and GXAircom cooperate again.

Did not know about the FANet topic...

rvt, 2025-06-06:
when I wrote the stratux compatibility I just wanted to know if it was GxAirCom, and if so simply configure GxAirCom.
I do believe it's important to stick to proper standards, so when no Obs is present in PFLAV it should not be send...

Create a custom sentence in GxAirCOm that stratux can detect (or others) sounds like to correct thing to do..

ABout FANET: From stratux perspective we just care about conspicuity, so re-using a HEX is indeed possible, but I don't think in real world that's going to be any issue at all. We have seen far to many near incidents and also some accidents to worry about that..

For the people that don't know: most of the time the HEX code is the same as the transponder code.

alxwolf, 2025-06-06:
@rvt agree, that would be clean. 

Now, we already have a `$PGXCF` message. And that is in the hands of GXAircom alone... 

Would it hurt to broadcast this config periodically? AFAIK Foreflight does that too, to announce its presence.

Probably would not hurt and provide a clear statement to Stratux (and others) about available features.

misterdoud, 2025-12-19:
it s very simple with the las stratux version 
make sure you have selected ont the web interface of GXaircom

output UPD
additional output over serial
output GPS		config GPS
output FLARM	
output FANET
output Vario LK8EX1

after you can check on the stratux interface and you can have a new modul fore setup the GXaircom