gereic/GXAirCom #203: Forwarding messages from mountain device to home ground station

alexzaganelli, 2025-07-04:
Hi GXAirCom team,

I’m planning to install a device on the mountain near my house to capture better signals, but there's no 3G/4G coverage there.

Two LoRa devices (mountain ↔ home) do communicate successfully. I'd like to know:

1. Is it possible to **forward the messages** received by the mountain device to my home device (configured as the ground station)?
2. Is this already supported (e.g., via the “broadcast-msg for fanet” feature introduced in v1.4.4)?
3. Or is forwarding not technically feasible with the current architecture?

Thanks in advance for your help.

Best regards,
Alex

gereic, 2025-07-14:
@mapettar 
This is not possible, because in the FLARM-Protocol a forward is not forseign.
Most of the devices today are sending FLARM and FANET, so I suggest to enable only FANET, otherwise you loose some packages, because it is not possible to listen to FLARM and FANET at the same time.

mapettar, 2025-07-14:
Hi,
I'm just interested in the Flarm protocol, and this is how it's configured.
Is it possible to install a ground station in the mountains where there's no Wi-Fi, connected to an esp32lora that transmits the signal to a lora receiver?
Basically, I need a Flarm receiver where I can't use Wi-Fi.

nepat, 2025-08-10:
Check whether the location truly doesn't have 4G NB-IOT connection. The SIM7080G wireless modem in the current board design supports the NB-IOT 4G mode. NB-IOT has very good range/signal penetration characteristics. It also uses very little power. Usually normal 'consumer type' sim cards will not support the NB-IOT protocol. It requires a sim card from a provider like 1NCE, Emnify, etc. that explicitely support NB-IOT ! Phones don't operate on NB-IOT. Testing with a normal phone will not tell you whether NB-IOT will work in the location.

gereic, 2025-09-02:
That can't work, cause this Board is not supported.

gereic, 2025-09-02:
The Lilygo Sim7000-Board should work, but you need an additional Board with the sx1276 Lira Module.
https://a.aliexpress.com/_EHLez3g
https://a.aliexpress.com/_EJmuRzy

mapettar, 2025-09-02:
> The Lilygo Sim7000-Board should work, but you need an additional Board with the sx1276 Lira Module. https://a.aliexpress.com/_EHLez3g https://a.aliexpress.com/_EJmuRzy
Thank you so much, that's the one I already ordered, I'll update you when it arrives.

mapettar, 2025-09-17:
Hi, SIM7000G arrived, firmware loaded and everything works fine, thanks you, left PayPal donation